ophub/amlogic-s9xxx-openwrt #132: s905X2 device not booted on B860H-V5 telkom

nationpwned, 2021-10-03:
hello ophub,
i have downloaded and burn s905x2 on my device b860h-V5 rooted but unluckily not booted,
i see in env.ext file the dtb use meson-g12a-x96-max.dtb

i try to browse and got info that this device use g12a_u212_2g
someone on indonesian's fb group provide the dts of this device, here is the link : https://www.mediafire.com/file/xnt75n4nqbfamn3/g12a_u212_2g.dts/file

and when i check that was true that this device use g12a_u212_2g
[image]

can u help me to make this device booted on openwrt?
thanks before

ophub, 2021-10-03:
Edit `/boot/uEnv.txt`, Try this dtb for 2G memory: `meson-g12a-x96-max-rmii.dtb`

Try it with the 5.4 kernel first: https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/openwrt_s9xxx_lede_2021.10.02.2109/openwrt_s905x2_k5.4.150_2021.10.02.2057.img.gz

wormalaska, 2021-10-03:
same case like nationpwned
[image]

so i tried https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/openwrt_s9xxx_lede_2021.10.02.2109/openwrt_s905x2_k5.4.150_2021.10.02.2057.img.gz

blackscreen only 
not showup

ophub, 2021-10-03:
I don’t know how your screen came from. How did you execute the ls command without starting?

You write the openwrt of the 5.4 kernel s905x2 into the USB disk, insert the USB disk into the box, and plug in the power to start.

It will start after a minute, connect the keyboard to your box, and press Enter to see the banner. Or you can directly visit the browser.

ophub, 2021-10-03:
Take a screenshot of /boot/uEnv.txt, take a screenshot to see the startup process, insert the U disk, power on, and take a screenshot of the startup screen one minute later. I don’t know what you are doing in the picture above.

nationpwned, 2021-10-03:
> Edit `/boot/uEnv.txt`, Try this dtb for 2G memory: `meson-g12a-x96-max-rmii.dtb`
> 
> Try it with the 5.4 kernel first: https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/openwrt_s9xxx_lede_2021.10.02.2109/openwrt_s905x2_k5.4.150_2021.10.02.2057.img.gz

none of g12a dtb work, 
i tried to change uboot with s905x2-s922 uboot renamed as uboot.ext also wont boot and enter to andorid tv.
i've tried 5.4 and 5.10 also wont boot

any idea how to make this device work?

nationpwned, 2021-10-03:
just info, b860h-v5 doesnt have reset button, what i did : adb connect 192.x.x.x.x, adb shell, su, reboot update.

ophub, 2021-10-03:
You do what I say, answer my question.

nationpwned, 2021-10-03:
> You do what I say, answer my question. I did not teach you these operations.
[image]

after reboot update, nothing happen except entering to android tv

ophub, 2021-10-03:
Do not change if you use the 5.4 kernel, except uEnv.txt, do not change anything else.

nationpwned, 2021-10-03:
> Do not change if you use the 5.4 kernel, except uEnv.txt, do not change anything else.

first time i tried is just change the dtb, after trying all g12a and none of those work, i tried change uboot, and now i just burn new image on my sd card..

ophub, 2021-10-03:
Insert the USB disk, press and hold the reset button, power on, release the reset button after 5 seconds, you can start from usb.

ophub, 2021-10-03:
Use usb disk, not tf card.

ophub, 2021-10-03:
5.4 The kernel has nothing to do with u-boot, as long as you choose the right dtb, you can use it.

nationpwned, 2021-10-03:
ok, i use usb disk, but the still entering to android tv,
i tried press 3 seconds, 5 seconds and more but still entering to android tv

ophub, 2021-10-03:
Try those few next to each other. `meson-g12a-*.dtb`

nationpwned, 2021-10-03:
> Try those few next to each other. `meson-g12a-*.dtb`

i have already done it, but no good result, still entering to android tv

ophub, 2021-10-03:
If your own dtb can't be started, you can go to the [armbian](https://forum.armbian.com) forum, find the s905x2 armbian system that suits your box, and copy his dtb to use.

ophub, 2021-10-03:
I don't think you have a dtb problem, but you have not booted from the USB flash drive, and you have entered the original Android tv system every time.

nationpwned, 2021-10-03:
> I don't think you have a dtb problem, but you have not booted from the USB flash drive, and you have entered the original Android tv system every time. You can boot from the USB disk first, and then try dtb.
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.md#1010-turn-on-the-usb-disk-boot-mode-of-the-amlogic-s9xxx-stb

ok, i will try to browe the entire net, and if this device work, i will give you info how this device work..
by the way thank you very much,
you did a brilliant job.
greeting from Indonesia

ophub, 2021-10-03:
You're welcome. There are many ways to set the box to boot from usb. As long as you enter the usb and start it, you are ready to debug.

closed 2021-10-06 by ophub