ophub/amlogic-s9xxx-openwrt #133: Box cannot boot via pendrive

kokenot, 2021-10-06:
is there any other way to boot the openwrt in the box? I have another x96 max plus that failed to boot openwrt via pendrive. model name show on setting android setting is x96 max plus a1. I'm using similar technique by pressing reset + power on, but still it cannot boot. this board version is different from my previous x96Max Plus. already change different firmware and different pendrive. no luck at all. the box only show recovery android if I insert hdmi to monitor. there is no sign of Black and White text booting process.

ophub, 2021-10-06:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.md#1010-turn-on-the-usb-disk-boot-mode-of-the-amlogic-s9xxx-stb

kokenot, 2021-10-06:
tq I will look into it

kokenot, 2021-10-06:
box can boot. problem the dtb is not compatible with the on board network cipset. this is the network cipset. other brand and not realtek. how can I get dtb version for this network cipset? or is there any trick to pull dtb file from stock firmware? 
[image]

ophub, 2021-10-06:
I did not find the introduction of `x96 max plus a1` equipment on the Internet. Can you see what model of its `soc` is?

closed 2021-10-08 by ophub