ophub/amlogic-s9xxx-openwrt #134: Kernel

OpenSourceop, 2021-10-07:
hi. how do you run loadbalance in your kernel? I tried your build mwan3 openwrt runs but I tried the armbian balbes kernel mwan3 can't run. thanks

ophub, 2021-10-07:
Used by flippy optimized configuration for aml s9xxx, he provided the [source code](https://github.com/unifreq/linux-5.4.y) and the compiled [kernel software package](https://github.com/ophub/flippy-kernel). You can go to his website for advice.

OpenSourceop, 2021-10-07:
> Used by `flippy` optimized configuration for aml s9xxx, he provided the [source code](https://github.com/unifreq/linux-5.4.y) and the compiled [kernel software package](https://github.com/ophub/flippy-kernel). You can go to his website for advice.

thanks for your reply i will try it. any other thing i want to know how to rename kernel in openwrt in luci view? example kernel name 5.10.66-flippy-64+ becomes 5.10.66

ophub, 2021-10-07:
It can be changed in `make menuconfig` —> `LOCALVERSION`.

You can also change it at compile time:
`export LOCALVERSION="-flippy-65+"`