ophub/amlogic-s9xxx-openwrt #138: X96 Max Plus Variant

farlexyz, 2021-10-09:
Hi, i found that X96 Max Plus have different variant while searching the internet. I have been trying to install using both method, emmc & pendrive, both failed to install because there is no ethernet driver. Kindly check the website to see X96 Max Plus variant:
https://4pda-to.translate.goog/forum/index.php?showtopic=1013103&st=40&_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=nui#entry102780026
My box currently have ICPlus_IP1001 for ethernet, i can be your tester if u want
[image]
[image]
.

ophub, 2021-10-10:
External `USB Wifi`, you can choose from several products from `ralink` and `medietek`, E.g:

RT2770, RT2870 & RT3070, RT3071 & RT3072,RT3370, RT3572,RT3573 ,RT5370, RT5572，mt7601u, mt7610u, mt7612u

`rt` is basically an 802.11n device, and `USB2.0` is available; 

`mt7610u` and mt7612u can support 802.11ac, and `USB3.0` is required.

！！！Realtek does not support at all.

ophub, 2021-10-10:
`X96 Max Plus `
Еще разновидность Plus_XKN Нет 1000Mb Ethernet, радиомодуль RTL8189FTV 802.11bgn 2.4G, память 2\16. Фото 

X96 Max Plus2. Наличие 1000Mb Ethernet RTL8211F. Удостовериться можно подключив в оборудование поддерживающие 1Gb или с помощью Device Info HW, или визуально по наличию чипа на плате. Антенн 2шт. Радиомодуль RTL8822CS
 
 `X96 Max Plus2 `
Появилась модификация Plus2, но 2GB оперативы, шьется только офпрошивка 202009 и спецпрошивка для 2GB RAM от JAW. Фото 
И еще есть версия Plus2 с радиомодулем RTL8822BS (Fn-Link 6222B-SRB), на портах югоса проблема с радиомодулем. 

X96 Max Plus100 Отсутствует чип 1000Mb Ethernet на плате. LAN 100, есть BT и WiFi 2G\5G (антенн две шт.), радиомодуль RTL8822CS

ophub, 2021-10-10:
Currently there is no driver for RTL8822CS on openwrt. Some boxes in the introduction you sent have 100M Ethernet ports, you can choose different firmware according to soc.

closed 2021-10-12 by ophub