ophub/amlogic-s9xxx-openwrt #139: Compile some package from external source

Manssizz, 2021-10-09:
how to compile package from external source? 
ie. ffmpeg from openwrt package have bug for s905x soc, i want use this [source](https://github.com/khadas/android_vendor_amlogic_external_ffmpeg) but idk how doing.

Manssizz, 2021-10-10:
How about this?
https://github.com/dlxtt/FFmpeg-aml

Manssizz, 2021-10-10:
temporary closed. because too hard and complicated

Manssizz, 2021-10-10:
i confuse because there have many different structure. external source have lot of folder and source didn't have release package, only source code.
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.md#example-1-add-a-third-party-software-package i was use this method for compiling few package like ser2net, replacing argon theme, etc. but back to point 1, external source have lot of folder and source didn't have release package, only source code.

anw... problem was sloved with docker. but if can build without docker, it really great.

ophub, 2021-10-10:
```
#The first step is to delete the current ffmpeg package
rm -rf feeds/packages/multimedia/ffmpeg

#The second step is to introduce the appropriate software package you found
svn checkout https://github.com/openwrt/packages/trunk/multimedia/ffmpeg feeds/packages/multimedia/ffmpeg
```

This completes the replacement of the software package. Take a look at which warehouse's ffmpeg can meet your needs, just replace it as long as the other party's source code supports use in openwrt.

ophub, 2021-10-10:
You said that the default ffmpeg has a bug, and how to replace it with others, so I teach you how to replace it.
I used the default and no problems were found.

Manssizz, 2021-10-11:
i was try replacing ffmpeg with openwrt source. if i run this script, i getting error 
`ffmpeg -i input.mkv -f srt -i subs.srt -map 0:0 -map 0:1 -map 1:0 -c:v copy -c:a copy -c:s srt  -metadata:s:s:0 title="Bahasa Indonesia" output.mkv`

i got error Decoding Time Stamp (DTS). it's make video glitch.
[image]