ophub/amlogic-s9xxx-openwrt #158: This step has been truncated due to its large size. Download the full logs from the  menu once the workflow run has completed.

thecupangin, 2021-11-05:
hi, i try to build with action on github, but failed and show error log "This step has been truncated due to its large size. Download the full logs from the  menu once the workflow run has completed." how to solve ya?
[image]

ophub, 2021-11-05:
```
Server space usage before starting to compile: 
Filesystem     Type  Size  Used Avail Use% Mounted on
/dev/root      ext4   84G   67G   17G  80% / 
```

The available space of github is `84G`. I used `67G` when compiling `openwrt-armvirt-64-default-rootfs.tar.gz`. If you choose more packages, you need more space. I suggest you run my default configuration first, and then add it gradually. Also leave enough space for the `packaged OpenWrt firmware` below.

thecupangin, 2021-11-05:
add more space? how to add more space?

ophub, 2021-11-05:
The free user of github is 84G, and the space for paid users is large.

ophub, 2021-11-05:
https://github.com/ophub/amlogic-s9xxx-openwrt/runs/4112221095?check_suite_focus=true#step:15:181

Enough for normal use.

thecupangin, 2021-11-05:
let me try your original script, use lede ya? not openwrt 21.02

thecupangin, 2021-11-05:
[image]
[image]

using default script, without adding new packages, still same issue

ophub, 2021-11-05:
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/1424485467

https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/1423854741

I tested both branches today, no problem. You synchronize, try to compile it first.

thecupangin, 2021-11-05:
[image]
 re-fork, fresh & up to date, hopefully work

thecupangin, 2021-11-05:
[image]
the problem still same, fresh fork, no edit

ophub, 2021-11-05:
You try to compile the lede branch.

thecupangin, 2021-11-05:
[image]
trying LEDE branch

thecupangin, 2021-11-05:
[image]
LEDE still same bro. what's wrong ya?

ophub, 2021-11-06:
https://www.google.com/search?q=github+This+step+has+been+truncated+due+to+its+large+size

I see that you have compiled it. Many records of this problem have been searched in google. Github’s prompts on the big log will continue to be executed.

closed 2021-11-06 by ophub

thecupangin, 2021-11-06:
why close?