ophub/amlogic-s9xxx-openwrt #162: workflows build fail!

kyliezero, 2021-11-13:
why does my built always fail? I've tried the build several times still the same, it's a waste of time, really really annoying! -_-
[image]

ophub, 2021-11-13:
I will compile it now and have a look.
https://github.com/ophub/amlogic-s9xxx-openwrt/actions

ophub, 2021-11-13:
Can compile normally

closed 2021-11-13 by ophub

kyliezero, 2021-11-13:
> I will compile it now and have a look. https://github.com/ophub/amlogic-s9xxx-openwrt/actions

I see it runs smoothly, is it possible that because I added diy-part2.sh there is a problem?
I tried to add the package openclash & passwall with this command into diy-part2.sh

`svn co https://github.com/xiaorouji/openwrt-passwall/trunk package/openwrt-passwall`
`svn co https://github.com/vernesong/OpenClash/trunk/luci-app-openclash package/openwrt-openclash
pushd package/openwrt-openclash/tools/po2lmo && make && sudo make install 2>/dev/null && popd`