ophub/amlogic-s9xxx-openwrt #165: change location temperature for 860v1 dtb

Manssizz, 2021-11-14:
hi... 
can u add conditional for pick temp location? 
because b860 dtb use this location for checking temp unit
`cpu_temp="$(awk '{ printf("%.1f °C", $0 / 1000) }' /sys/class/hwmon/hwmon0/temp1_input)"`
for now, i use manual changing with mark your script as comment and add these script. 
thx!

ophub, 2021-11-15:
Where to display, screenshot
```
cat /boot/uEnv.txt
cat /etc/flippy-openwrt-release
echo $(awk '{ printf("%.1f °C", $0 / 1000) }' /sys/class/hwmon/hwmon0/temp1_input)
echo $(sensors | grep -Eo '\+[0-9]+.+C' | sed ':a;N;$!ba;s/\n/ /g;s/+//g')
echo $(awk '{ printf("%.1f °C", $0 / 1000) }' /sys/class/thermal/thermal_zone0/temp)
```
The above commands are executed one by one, and the screenshots are given to me.

Manssizz, 2021-11-15:
Ah sorry i forgot write file location
On `/etc/profile.d/30-sysinfo.sh` have line code with `cpu_temp="$(awk '{ printf("%.1f °C", $0 / 1000) }' 
/sys/class/thermal/thermal_zone0/temp)"`
I replace with
`cpu_temp="$(awk '{ printf("%.1f °C", $0 / 1000) }' /sys/class/hwmon/hwmon0/temp1_input)"`

[image]

ophub, 2021-11-15:
0fc863a

You try, feedback the result.

Manssizz, 2021-11-15:
it works! thank you
[image]

closed 2021-11-16 by ophub