ophub/amlogic-s9xxx-openwrt #166: Luci fails to start, openwrt_s905x3_k5.4.158_2021.11.12.1717

MartyOFaherty, 2021-11-15:
Fresh install of OpenWRT 21.02.1 on an X96 Air Android TV box. OpenWRT running on USB stick. Accessing the web interface gives a Bad Gateway error.

Running /www/cgi-bin-luci over ssh results in the following:

root@OpenWrt:~# /www/cgi-bin/luci /usr/bin/lua: /www/cgi-bin/luci:4: attempt to index field 'dispatcher' (a nil value)
stack traceback: /www/cgi-bin/luci:4: in main chunk
[C]: ?

The luci script is simply:

root@OpenWrt:~# cat /www/cgi-bin/luci
 #!/usr/bin/lua
 require "luci.cacheloader"
 require "luci.sgi.cgi"
 luci.dispatcher.indexcache = "/tmp/luci-indexcache"
 luci.sgi.cgi.run()

I've tried creating an empty directory /tmp/luci-indexcache with no difference to the result. Deleting it again also makes no difference.

I suspect there may be an issue with luci.cacheloader, but I don't know where to find it, or whether it is text or binary. I have NO lua experience. Googling doesn't return anything that looks like a solution.

ophub, 2021-11-15:
```
rm -rf /tmp/luci-*
reboot
```

MartyOFaherty, 2021-11-15:
Ran rm command.
Ran ls to confirm deletion. /tmp/luci-modulecache definitely deleted.
Rebooted.

Attempted to access web interface, got "Bad Gateway" again.

Ran /www/cgi-bin-luci over ssh again with the same result.

/tmp/luci-modulecache directory has been reinstated.

ophub, 2021-11-15:
I tested it and it is normal, you can download and install again

[image]

closed 2021-11-15 by ophub