ophub/amlogic-s9xxx-openwrt #167: Install To EMMC

adisubagja, 2021-11-19:
I have two device b860h and then have different model emmc
first i trying install on emmc ( Model MMC DG4008 ) This works as well perfectly but on emmc ( Model MMC 8GTF4R ) Installed but can't boot ? why any solution for this ?

ophub, 2021-11-19:
It is better to use on USB, with more space and complete functions.

adisubagja, 2021-11-19:
[image]
Yes but i want to intall to emmc this screenshoot done installed but can't boot

ophub, 2021-11-20:
Connect the monitor to see what prompt message is.

adisubagja, 2021-11-20:
Hmmm ok,
Thanks will try again im must buy monitor first :D

closed 2021-11-20 by adisubagja