ophub/amlogic-s9xxx-openwrt #171: Support for "s905"?

carlocensi813, 2021-11-21:
Hello, I'm trying to boot your openwrt in my tv box without any luck, I just keep getting reboot 1 minute after boot starts to load stuff.

My tv box is beelink mini mx 1gb/8gb gxbb p200(I did config that DTV on uEnv.txt file)

Im using your s905x 5.4 IMG, because couldn't find an s905 one.

 Thanks

ophub, 2021-11-22:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-dtb

There are many dtb files of meson-gxbb_p200_xxxx.dtb, you can try.

Modify location: 
`/boot/uEnv.txt`
`FDT=your_dtb`

closed 2021-11-22 by ophub

carlocensi813, 2021-11-22:
I tried with the right dtb for my device, which is "meson_gxbb_p200_1G_1Gbit.dtb" but I just got a blackscreen and nothing happens, no reboot after loading some code like last time, just a blackscreen. Should I try with another IMG? Im using s905x, kernel 5.4

carlocensi813, 2021-11-22:
> https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-dtb
> 
> There are many dtb files of meson-gxbb_p200_xxxx.dtb, you can try.
> 
> Modify location: `/boot/uEnv.txt` `FDT=your_dtb`

I tried with the right dtb for my device, which is "meson_gxbb_p200_1G_1Gbit.dtb" but I just got a blackscreen and nothing happens, no reboot after loading some code like last time, just a blackscreen. Should I try with another IMG? Im using s905x, kernel 5.4