ophub/amlogic-s9xxx-openwrt #172: S905 support

carlocensi813, 2021-11-23:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/amlogic-dtb

There are many dtb files of meson-gxbb_p200_xxxx.dtb, you can try.

Modify location: `/boot/uEnv.txt`
`FDT=your_dtb`

_Originally posted by @ophub in https://github.com/ophub/amlogic-s9xxx-openwrt/issues/171#issuecomment-974996610_

I tried with the right dtb for my device, which is "meson_gxbb_p200_1G_1Gbit.dtb" but I just got a blackscreen and nothing happens, no reboot after loading some code like last time, just a blackscreen. Should I try with another IMG? Im using IMG for s905x, kernel 5.4

ophub, 2021-11-24:
```
meson-gxbb-p200.dtb
meson-gxbb-p201.dtb
meson-gxbb_p200.dtb
meson-gxbb_p201.dtb
```
Under one test, what device are you? 100M network card?

carlocensi813, 2021-11-24:
This is the right dtb "meson_gxbb_p200_1G_1Gbit.dtb" is 1000Mb network card(1gbit) but it gives me just blackscreen, never boots. this one "meson-gxbb-p200.dtb" doesn't work it boots but reboot on a loop after one minute.

[https://coreelec.org/dtb/](url) <html>

As you can see there my device is:

Beelink Mini MX | S905 | 1G | gxbb_p200_1G_1Gbit

carlocensi813, 2021-11-24:
> ```
> meson-gxbb-p200.dtb
> meson-gxbb-p201.dtb
> meson-gxbb_p200.dtb
> meson-gxbb_p201.dtb
> ```
> 
> Under one test, what device are you? 100M network card?

This is the right dtb "meson_gxbb_p200_1G_1Gbit.dtb" is 1000Mb network card(1gbit) but it gives me just blackscreen, never boots. this one "meson-gxbb-p200.dtb" doesn't work it boots but reboot on a loop after one minute.

This is the right dtb "meson_gxbb_p200_1G_1Gbit.dtb" is 1000Mb network card(1gbit) but it gives me just blackscreen, never boots. this one "meson-gxbb-p200.dtb" doesn't work it boots but reboot on a loop after one minute.

https://coreelec.org/dtb/

As you can see there my device is:

Beelink Mini MX | S905 | 1G | gxbb_p200_1G_1Gbit

carlocensi813, 2021-11-24:
> ```
> meson-gxbb-p200.dtb
> meson-gxbb-p201.dtb
> meson-gxbb_p200.dtb
> meson-gxbb_p201.dtb
> ```
> 
> Under one test, what device are you? 100M network card?

This is the right dtb "meson_gxbb_p200_1G_1Gbit.dtb" is 1000Mb network card(1gbit) but it gives me just blackscreen, never boots. this one "meson-gxbb-p200.dtb" doesn't work it boots but reboot on a loop after one minute.

https://coreelec.org/dtb/

As you can see there my device is:

Beelink Mini MX | S905 | 1G | gxbb_p200_1G_1Gbit

ophub, 2021-11-24:
There are currently no user usage reports, so I can only test it myself.

closed 2021-11-24 by ophub