ophub/amlogic-s9xxx-openwrt #173: 不能从SD卡启动5.10系统

szhhq, 2021-11-25:
我的章鱼星球已经刷了F大的固件，内核是5.4的。
我在SD卡上写入这里发布的内核5.4的固件，开机可以启动SD卡上的系统；但是SD上写入内核5.10的固件，然后启动的还是机器原来的F大5.4固件。
不知道是怎么回事？

ophub, 2021-11-25:
The difference between 5.4 and 5.10 is the kernel, you can upgrade the kernel to 5.10 after installing 5.4.
Log in to `OpenWrt`, `system menu`, `Amlogic service`, select 5.10 in the `settings`, and `update` the kernel online

szhhq, 2021-11-25:
谢谢你的及时回复！
另外的问题：
我从SD上启动系统后，从系统软件仓库安装iptables-mod-tproxy失败。
上传ipk后用-force-depends强制安装能安装上，但自动下载安装的很多依赖的版本号是5.10的，可是系统内核是5.4的呀。这是更新源的问题吗？
我在已关闭的问题中看到N1盒子也有类似的问题。如果说是需要在编译固件时就包含进去，那以后想要安排别的软件可能还会有类似的问题呀。

ophub, 2021-11-25:
The box is not a device officially supported by OpenWrt by default, so some packages cannot be installed from official sources.
Compiling and integrating the software packages you need is a better way.

closed 2021-11-26 by ophub