ophub/amlogic-s9xxx-openwrt #174: 大神能否出个带ssr的章鱼67+0

yu13740000, 2021-11-27:
装了n多章鱼固件，网页测速和手机speedtest都不能满速，网管交换机单臂路由模式。今天刷了一下你的 openwrt_s912_k5.4.162_2021.11.26.2052.img.gz速度出奇的满速，就是连个ssr都没有，安装缺少太多依赖，安装不上，能否出个带ssr的固件，或者编译ssr的依赖，菜鸟只会简单刷机，谢谢了。

yu13740000, 2021-11-27:
章鱼近两年的固件，网页测速基本都不能满速，你这个章鱼67+0测速满速太意外了，67+也测试过也不能满速。

ophub, 2021-11-28:
Are you looking for this? https://github.com/ophub/amlogic-s9xxx-openwrt/commit/caa6443c8c7154e278185912456237bb8b844f5c

I wrote a method for customizing openwrt firmware. With a very simple operation, you can get a firmware that suits you. Some personalized software packages can be added/deleted/modified by yourself. The method is as follows:

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.cn.md#%E4%B8%BE%E4%BE%8B1%E6%B7%BB%E5%8A%A0%E7%AC%AC%E4%B8%89%E6%96%B9%E8%BD%AF%E4%BB%B6%E5%8C%85

closed 2021-11-28 by ophub