ophub/amlogic-s9xxx-openwrt #175: Support device X96Max Plus2 T

adisik, 2021-11-27:
I have X96Max Plus2 T with 1000М Ethernet RTL8211F chip and WIFI AM7256. 
Is it possible to install an Openwrt on this hardware?

ophub, 2021-11-28:
https://github.com/ophub/amlogic-s9xxx-openwrt/issues/138

RTL8211F can use.

adisik, 2021-11-28:
My board

[image]

adisik, 2021-11-28:
Messages, when i try boot from USB flash with openwrt image:
[image]
[image]

adisik, 2021-11-28:
@ophub , As I understand it, Openwrt not loaded correct?

ophub, 2021-11-28:
This is the start, right? Connect the keyboard and press Enter to see the banner.

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.md#1010-turn-on-the-usb-disk-boot-mode-of-the-amlogic-s9xxx-stb

closed 2021-11-29 by ophub

test012012, 2022-02-11:
Hi @adisik .I have the same board. Could you please describe which image you use and how you boot Armbian or openwrt. Do you boot (first time) from SD card or from USB flash/hard disk?

I can see only black screen after boot logo