ophub/amlogic-s9xxx-openwrt #182: spam log thermal monitoring luci uhttpd

Manssizz, 2021-12-30:
- Device: b860h v1
- Case: syslog full spammed with thermal monitoring No such file or directory
- Linked issue: https://github.com/ophub/amlogic-s9xxx-openwrt/issues/165
- Sample proof:
```
Thu Dec 30 12:57:01 2021 daemon.err uhttpd[2855]: awk: /sys/class/thermal/thermal_zone0/temp: No such file or directory
Thu Dec 30 12:57:08 2021 daemon.err uhttpd[2855]: awk: /sys/class/thermal/thermal_zone0/temp: No such file or directory
Thu Dec 30 12:57:13 2021 daemon.err uhttpd[2855]: awk: /sys/class/thermal/thermal_zone0/temp: No such file or directory
Thu Dec 30 12:57:18 2021 daemon.err uhttpd[2855]: awk: /sys/class/thermal/thermal_zone0/temp: No such file or directory
Thu Dec 30 12:57:23 2021 daemon.err uhttpd[2855]: awk: /sys/class/thermal/thermal_zone0/temp: No such file or directory
Thu Dec 30 12:57:29 2021 daemon.err uhttpd[2855]: awk: /sys/class/thermal/thermal_zone0/temp: No such file or directory
Thu Dec 30 12:57:34 2021 daemon.err uhttpd[2855]: awk: /sys/class/thermal/thermal_zone0/temp: No such file or directory
Thu Dec 30 12:57:40 2021 daemon.err uhttpd[2855]: awk: /sys/class/thermal/thermal_zone0/temp: No such file or directory
```
[image]

ophub, 2021-12-30:
In the system-ttyd terminal, or ssh into openwrt, enter the following command, and give me a screenshot.

`sed -n 130,140p /etc/profile.d/30-sysinfo.sh`

ophub, 2021-12-30:
If your version is one month ago, you can update to the latest.

```
#Download the latest file
wget -O /etc/profile.d/30-sysinfo.sh https://raw.githubusercontent.com/ophub/amlogic-s9xxx-openwrt/main/amlogic-s9xxx/common-files/patches/cpustat/30-sysinfo.sh

#Grant permissions (can be ignored)
chmod +x /etc/profile.d/30-sysinfo.sh

reboot
```

Manssizz, 2021-12-30:
use latest version of 30-sysinfo.sh based you repos
[image]

ophub, 2021-12-30:
```
elif [ -f "/sys/class/thermal/thermal_zone0/temp" ]; then
	cpu_temp="$(awk '{ printf("%.1f °C", $0 / 1000) }' /sys/class/thermal/thermal_zone0/temp)"
```
30-sysinfo.sh: delete line 133-134

closed 2021-12-31 by ophub

Manssizz, 2021-12-31:
still shown btw
[image]

seems this issue linked with uhttpd scripts
[image]