ophub/amlogic-s9xxx-openwrt #183: Compilation failed

riyuejz, 2021-12-31:
hi can you help me to add luci-app-openclash into build openwrt-21.02, i tried by adding line

#diy-part2.sh
`svn co https://github.com/vernesong/OpenClash/trunk/luci-app-openclash package/openwrt-openclash
pushd package/openwrt-openclash/tools/po2lmo && make && sudo make install 2>/dev/null && popd`

#config
`CONFIG_PACKAGE_luci-app-openclash=y`

but compiling results always fails

ophub, 2022-01-01:
Your approach is correct. What error was reported? Send a screenshot.

riyuejz, 2022-01-01:
The logs are too big to read, it's hard to know where the error is, but after adding this package, the compilation failed
[image]

ophub, 2022-01-01:
You can look forward to find out if there are any errors.

Yours is a problem that you may encounter when compiling openwrt-21.02, and github will get stuck and report an error in the long log.

No need to change anything, do it again.

ophub, 2022-01-01:
I see your repo has been successful

closed 2022-01-01 by ophub