ophub/amlogic-s9xxx-openwrt #184: how to rebuild uInitrd/initrd.img

0neday, 2022-01-07:
> I am trying to build kernel and ulnitrd, even uboot.

I use [Amlogic_s905-kernel](https://github.com/povlhp/Amlogic_s905-kernel), select kernel version `5.0.2` and  build successfully.
and use [this article](http://h-wrt.com/en/mini-how-to/uInitrd) to extract `initrd.img-5.4.170-flippy-68+o` file which come from Flippy , and check file directory, and find, which need kernel modules `/lib/modules/<kernel_version>`.
so, I need to replace modules in `uInitrd.img-5.4.170-xxx` using 5.0.2.

ophub, 2022-01-07:
He is the kernel source code and also open source: https://github.com/unifreq

For the compilation method of the custom kernel, see [compile-kernel](https://github.com/ophub/amlogic-s9xxx-openwrt#compile-a-custom-kernel)

closed 2022-01-07 by ophub

0neday, 2022-01-07:
Thank you, I have check [script](https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/compile-kernel/tools/script/generate_uinitrd.sh). but how to create build environment in x86 hosts?

ophub, 2022-01-08:
https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/compile-kernel

You don’t need to change anything, just use it directly under x86. Read the instructions several times. There is an introduction inside.

Install the necessary packages (E.g Ubuntu 20.04 LTS user)
```
sudo apt-get update -y
sudo apt-get full-upgrade -y
sudo apt-get install -y $(curl -fsSL git.io/ubuntu-2004-server)
```