ophub/amlogic-s9xxx-openwrt #185: Do not boot from usb after using dd command to write rootfs and boot parition

0neday, 2022-01-09:
I use 
```
dd if=/dev/sda1 of=/dev/mmcblk2p1
dd if=/dev/sda2 of=/dev/mmcblk2p2
```
after that, I never can boot from usb. 
so, maybe is uboot problem?
thank you

ophub, 2022-01-09:
Then write the IMG file to the USB hard disk through software such as `Rufus` or `balenaEtcher`.

closed 2022-01-09 by ophub

0neday, 2022-01-09:
> Then write the IMG file to the USB hard disk through software such as `Rufus` or `balenaEtcher`.

not this means.
I use rufus write img to usb, and then plug usb into n1, then use 

> dd if=/dev/sda1 of=/dev/mmcblk2p1
> dd if=/dev/sda2 of=/dev/mmcblk2p2

write into emmc.

but after that, I can not boot from usb again!

ophub, 2022-01-09:
Install OpenWrt

Log in to the default IP: 192.168.1.1 → Login in to openwrt → system menu → Amlogic Service → Install OpenWrt

0neday, 2022-01-09:
yes, I use offical openwrt-defualt-armvirt-rootfs.tar.gz, so that I can use all packages exclude kmod packages.
I not use your rootfs, so I can not use your method to write partitions into emmc from usb.

0neday, 2022-01-09:
RT，
1.先上效果图。
[image]
[image]

2.具体改动的地方。

内核及模块，根据这个[github里的资源自己编译](https://github.com/ophub/amlogic-s9xxx-openwrt)，root-default系统用的[openwrt自带的](https://downloads.openwrt.org/releases/21.02.1/targets/armvirt/64/openwrt-21.02.1-armvirt-64-default-rootfs.tar.gz)。

ipv6
主要改动的地方
lan MTU，1500
pppoe mtu 1492
手动输入
[image]
![微信图片_1120220109194110.png]（https://s2.loli.net/2022/01/09/Wyh8U1GSQDKc57x.png）

只用salcc协议，获取ipv6 client ip
[image]

ophub, 2022-01-09:
推荐你自定义编译openwrt-defualt-armvirt-rootfs.tar.gz，而不是使用官方的。里面缺几个必选的工具包，如uuid，fdisk等，这些是你将openwrt从U盘写入emmc所需的依赖。我的仓库里route-config里有lede和官方的两个配置文件，必须的我都选好了，你可以在此基础上添加各种插件。这样的固件才适合盒子。

0neday, 2022-01-09:
I have restored img from ramdisk, thank you