ophub/amlogic-s9xxx-openwrt #187: dnsmasq-full replaces dnsmasq

hailong259, 2022-01-12:
I have configured the .config file as follows:

> # CONFIG_PACKAGE_dnsmasq is not set
> # CONFIG_PACKAGE_dnsmasq-dhcpv6 is not set
> CONFIG_PACKAGE_dnsmasq-full=y
> CONFIG_PACKAGE_dnsmasq_full_dhcp=y
> # CONFIG_PACKAGE_dnsmasq_full_dhcpv6 is not set
> # CONFIG_PACKAGE_dnsmasq_full_dnssec is not set
> # CONFIG_PACKAGE_dnsmasq_full_auth is not set
> CONFIG_PACKAGE_dnsmasq_full_ipset=y
> # CONFIG_PACKAGE_dnsmasq_full_conntrack is not set
> # CONFIG_PACKAGE_dnsmasq_full_noid is not set
> # CONFIG_PACKAGE_dnsmasq_full_broken_rtc is not set
> CONFIG_PACKAGE_dnsmasq_full_tftp=y

but when I build I get an error like this:

> 2022-01-12T05:20:17.1502503Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1504397Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/uci-defaults/50-dnsmasq-migrate-resolv-conf-auto.sh
> 2022-01-12T05:20:17.1505963Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1517267Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/lib/dnsmasq/dhcp-script.sh
> 2022-01-12T05:20:17.1518575Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1520034Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/sbin/dnsmasq
> 2022-01-12T05:20:17.1521197Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1522694Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/share/acl.d/dnsmasq_acl.json
> 2022-01-12T05:20:17.1523932Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1525557Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/share/dnsmasq/dhcpbogushostname.conf
> 2022-01-12T05:20:17.1526903Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1528408Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/share/dnsmasq/rfc6761.conf
> 2022-01-12T05:20:17.1529620Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1530524Z  * opkg_install_cmd: Cannot install package dnsmasq-full.
> 2022-01-12T05:20:17.1532222Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/hotplug.d/ntp/25-dnsmasqsec
> 2022-01-12T05:20:17.1533781Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1535260Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/init.d/dnsmasq
> 2022-01-12T05:20:17.1536432Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1538285Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/uci-defaults/50-dnsmasq-migrate-resolv-conf-auto.sh
> 2022-01-12T05:20:17.1539957Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1541491Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/lib/dnsmasq/dhcp-script.sh
> 2022-01-12T05:20:17.1542730Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1544171Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/sbin/dnsmasq
> 2022-01-12T05:20:17.1545341Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1546830Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/share/acl.d/dnsmasq_acl.json
> 2022-01-12T05:20:17.1548061Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1549772Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/share/dnsmasq/dhcpbogushostname.conf
> 2022-01-12T05:20:17.1550925Z 	But that file is already provided by package  * dnsmasq
> 2022-01-12T05:20:17.1552229Z  * check_data_file_clashes: Package dnsmasq-full wants to install file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/usr/share/dnsmasq/rfc6761.conf
> 2022-01-12T05:20:17.1553288Z 	But that file is already provided by package  * dnsmasq

Please help me, thanks you!

ophub, 2022-01-12:
This component is not present in the default config. I don't understand it.

closed 2022-01-12 by ophub

hailong259, 2022-01-12:
thank you for the quick response!

0neday, 2022-01-13:
Not need to do this, use dnsmasq defualt.
After install system. use openwrt offical source to install dnsmasq-full and remove dnsmasq, and then, use ln -sf to link /lib/modules/flip<kernel_vermagic>/*.ko