ophub/amlogic-s9xxx-openwrt #189: Boot from usb, but initrd mount /dev/mmcblk2p2, rather than /dev/sda2

0neday, 2022-01-15:
I flash your latest firmware  `openwrt_s905d_k5.4.171_2022.01.15.1130.img` , and then plug into usb port. 

I found, maybe boot from usb work, but initrd mount /dev/mmcblk2p2
```
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
udev                    512.0K         0    512.0K   0% /dev
tmpfs                   181.0M    108.0K    180.9M   0% /run
/dev/mmcblk2p2          973.0M    199.5M    767.7M  21% /
tmpfs                   905.1M     84.0K    905.0M   0% /tmp
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/mmcblk2p4            4.5G    161.6M      4.1G   4% /mnt

root@OpenWrt:/opt# uname  -a
Linux OpenWrt 5.4.171-flippy-69+o #366 SMP PREEMPT Tue Jan 11 23:58:30 CST 2022 aarch64 GNU/Linux

root@OpenWrt:/opt# ls /lib/modules/5.4.1
5.4.154/             5.4.170-meson64-dev/
root@OpenWrt:/opt# lsmod
gpio_keys_polled       16384  0

```
this show ramdisk init kernel and load drivers for hardwares, and then mount rootfs is `/dev/mmcblk2p4`, rather than `/dev/sda2`
this so wierd!

ophub, 2022-01-15:
You are not booting from usb. The system is still booted from emmc.

Re-download the firmware, re-write the U-disk, insert the U-disk into the box, and then re-plug the power supply to boot from the U-disk.

0neday, 2022-01-15:
> What does extracting initrd.img do? Compile the kernel?

nop,  check why mount /dev/mmcblk2p2, rather than /dev/sda2, 

check `cat /proc/cmdline` get some bootargs, 
```
BOOT=ROOTFS
```
from `. /scripts/local` function ``local_mount_root` 

I dont know how initrd get rootfs uuid? 

maybe I need a console to get into `init ramdisk` when system boot.

ophub, 2022-01-15:
As long as boot from usb, sda is automatically mounted. Your current situation shows that you have not booted from usb, some n1 boxes choose a U disk, change a U disk, or change another socket, change the firmware.

0neday, 2022-01-15:
I confirm, usb has been read,

[image]

ophub, 2022-01-15:
But the firmware in the U disk is invalid.

0neday, 2022-01-15:
> But the firmware in the U disk is invalid.

is valid, I confirm!

I can mount `/dev/sda2` and `/dev/sda1` on current system.

0neday, 2022-01-15:
I am consider use `panic ` in ramdisk when hit `ENTER keyboard` when boot, So, I can get a shell console. 
I have check `panic ` functions 
```
		# We don't have 'setsid -c' so we need to setsid, open
		# the tty, and finally exec an interactive shell
		REASON="$*" PS1='(initramfs) ' setsid sh -c "exec sh -i <>/dev/${console} 1>&0 2>&1"
	else
		REASON="$*" PS1='(initramfs) ' sh -i </dev/console >/dev/console 2>&1
	fi
```
there is a method is rename `/init` and `/sbin/init ` on current rootfs, so that produce panic 
```
# Check init is really there
if ! validate_init "$init"; then
	echo "Target filesystem doesn't have requested ${init}."
	init=
	for inittest in /sbin/init /etc/init /bin/init /bin/sh; do
		if validate_init "${inittest}"; then
			init="$inittest"

			break
		fi
	done
fi

# No init on rootmount
if ! validate_init "${init}" ; then
	panic "No init found. Try passing init= bootarg."
fi
```

ophub, 2022-01-15:
Your U disk is manually mounted for storage. The firmware in your USB flash drive is invalid, and the boot from the USB flash drive is not properly booted.

closed 2022-01-15 by ophub

0neday, 2022-01-16:
@ophub 
I flash  [flippy](https://t.me/openwrt_flippy/758)  `重制版_openwrt_s905d_n1_R21.6.1_k5.4.123-flippy-59+o.7z`, and boot usb successfully. 

so, I will compare your Initrd and flippy, when I get result , I will create PR

0neday, 2022-01-16:
@ophub 
use `blkid ` get [ your firmware ](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/openwrt_s9xxx_lede_2022.01.15.1143/openwrt_s905d_k5.4.171_2022.01.15.1130.img.gz),
```
xx-pc@xx-virtual-machine:~/Desktop$ sudo blkid  /dev/loop3p2
/dev/loop3p2: LABEL="ROOTFS" UUID="b250e917-e0a6-4061-9145-b8ee0cc20fea" UUID_SUB="fdb0d788-5b35-4d1e-8619-75a37574ddba" TYPE="btrfs"
```
and then flush your firmware into usb, then change 
```
sed -i "s/LABEL=ROOTFS/UUID=b250e917-e0a6-4061-9145-b8ee0cc20fea/" uEnv.txt
```
plug usb into N1 box, and boot successfully

[image]