ophub/amlogic-s9xxx-openwrt #196: Question for openwrt on S9xx devices

xtendpayaa, 2022-01-19:
Hi all!
Sorry to open an issue here, but I found your GitHub page for S9XX devices and I have lot of questions. Do you have a forum or IRC channel or something else where I can ask about it?
Thank you!

ophub, 2022-01-20:
Hello, if you have any questions, you can communicate here

xtendpayaa, 2022-01-20:
Hi and thank you!
So what is in my mind:
I'm thinking to buy a cheap s905x3 tv box and make an openwrt router from it with a usb3 ethernet dongle and for wlan need to use dumb access points connected via cable to the box.
I want to run rclone mount in the box and share via NFS/samba with multimedia devices. I'm already tried it with a low-end router but the router CPU was weak for that job because with 4K movie streaming the dual core 770Mhz CPU used 100% but the memory usage was very low.

So now I have in home an X92 S912 tv box with CoreElec. I try to mount Gdrive on CE succesfully and when I play 4K videos the CPU usage go up around 30-40% but only when fill up the buffer after going down about 10%, plus the box running the OS + Kodi + playing the movie so it's very good, it can handle more 4K streamings.

So this is why I'm thinking to buy an s905x3 because it has usb3 port so I can use usb3 ethernet dongle to share the internet with the dumb AP. 
But I know that box can use the gigabit bandwith with iperf3 test but when need to act like a router (start NAT jobs too) it is working good? 
What is the speeds with box if you use like a router?
What is CPU usage when it's act like a router? Is it stable?

ophub, 2022-01-20:
You can consider hk1, gigabit can be at full speed, and the CPU consumes 50% when watching 4K video. The operation is relatively stable. You can replace it with an acrylic case and add a cooling fan.

https://www.youtube.com/watch?v=eKgF1A5FMH8

https://www.youtube.com/watch?v=RYRd_M_e9NQ

xtendpayaa, 2022-01-20:
Thank you!
So the Hk1 box with your firmware can handle NAT, dhcp operations easily?
Am I thinking good that i can connect a cheap wifi router via usb3 ethernet dongle so the cheap router can "spread" WiFi in the house?

I tried yesterday your firmware (kernel 5.4) in my s912. But LuCI in Chinese language.
Today I plug the SD card again but got only black screen. Need to flash SD card again? And what need to copy and where if I want use your newer kernel firmware? Thank you

ophub, 2022-01-20:
System -> Language and Interface
Language can be modified. The default is only English and Chinese. If you use other languages, you can modify the .config file first and compile the firmware containing your own language.

Different socs have different firmware, the box of s905x3 can not be used with s912.

Compilation documentation: https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.md

xtendpayaa, 2022-01-20:
Thank you! Of yourse I know that different socs different fw. I'm using now s912 5.4 firmware with q201 dtb file. I don't know why I see now only black screen but the default OpenWRT wifi came up so I can connect to LuCI. In there the default language is Chinese but I can manage to change to english.
1 .But I'm connect the tv box to my router 5G router with ethernet cable but don't got internet via wifi or wlan. Even I can't make a opkg update because no internet. I booted back to android and there is internet with the cable. I don't know what is the problem.
2. i see there is docker pre-installed. But it's not a must have thing? I mean I can switch it off for save performance?
Sorry to ask a lot but I'm practice with my S912 box that my HK-1 box arrive I can set it up fully.

ophub, 2022-01-21:
Docker and other software packages can be customized to increase or decrease firmware by modifying .config.

There is no difference between using OpenWrt in the box and using it in routers such as x86_64. How you choose a custom software package and how to configure luci is exactly the same.

xtendpayaa, 2022-01-21:
Thank you!
I think the problem is the dtb file.
My device is X92 3GB/16GB 1gigabit lan. The recovery says q201. But for example with CoreElec need to use special dtb what name is: 
X92 | S912 | 3G (q201) | gxm_q201_3g_1gbit.dtb
Because there are 3 X92 device in the business 2/3GB RAM with 100M ethernet: gxm_q200.dtb and the 1000M ethernet is gxm_q201
But in your fw there is only meson-gxm-q201.dtb. The device boot up with it and I have internet with usb3 dongle but the box ethernet is dead. So I can't connect it the AP router.
I'm using now: openwrt_s912_k5.4.172_2022.01.17.2333.img.gz and with SD card.

xtendpayaa, 2022-01-21:
Thank you!
I just found that page! But strange for me that q201_3g, neither q200_3g dtb file not working stuck on X92 logo screen.

Now I changed dtb to meson-gxm-q200.dtb and have luck, the box ethernet is alive now, but showing only 100 mb/s half duplex. The usb ethernet dongle shows 1000mb/s full-duplex.
I found this info here but for different box: "When used in USB, only 100M Ethernet is supported, and Gigabit is supported after writing into EMMC." --- is that true? So in ssh openwrt-install-amlogic and give meson-gxm-q200.dtb ?

xtendpayaa, 2022-01-21:
The system is in now in EMMC but still only 100M the box ethernet so I think still some dtb problems. I'm little bit confused.
[image]

ophub, 2022-01-21:
try meson-gxm-octopus-planet.dtb on usb

xtendpayaa, 2022-01-21:
Yepp! You are the man!
Now I'm on EMMC with octopus and connected to my dumb AP everything is fine!
[image]

ophub, 2022-01-21:
try `openwrt_s912_k5.15.15`_2022.01.17.2334.img.gz on usb.
Don't change anything, just default.

xtendpayaa, 2022-01-21:
Sorry but not happening nothing because now the 5.4kernel fw in the EMMC, so I plug the 5.15.15 SD card but the 5.4 boots up.
I think need to flash back first an android :)
What openwrt packages using your project? I mean I tried install NFS but no package found.
root@OpenWrt:~# opkg install nfs-kernel-server
Unknown package 'nfs-kernel-server'.

And I see some signature check failed text then opkg update is running.
Now I'm flashing back some android and coming back.

ophub, 2022-01-21:
You don't need to flash back to Android, just flash the U disk again, usually because the U disk is not well written.

xtendpayaa, 2022-01-21:
Sorry but I flashed it 3x and always the 5.4 kernel version (what is in the emmc) loading.

"Update the software source and replace it with your domestic mirror address" -- how can I do that?

ophub, 2022-01-21:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.md#106-opkg-package-manager

https://openwrt.org/docs/guide-user/additional-software/opkg#adjust_repositories

xtendpayaa, 2022-01-21:
Thank you very much for all the help!

ophub, 2022-01-21:
https://www.geekbuying.com/item/X92-Amlogic-S912-Android-6-0-3G-16G-4K-60FPS-TV-BOX--379064-379064.html

Is this your box?

xtendpayaa, 2022-01-21:
Yes! This is my box!
Is that possible to use the latest openwrt and all that packages?

ophub, 2022-01-21:
yes

xtendpayaa, 2022-01-21:
I put back the original firmware of the X92 box and flash the openwrt_s912_k5.4.173_2022.01.21.1831.img.gz with q201.dtb and working successfully!
I tried with default dtb and q201 dtb the newer kernel but not booted, I don't know why.

Before that k5.4.173 I used the k5.4.172 and tried to update the kernel with LuCI amlogic service but after reboot I got million errors. Is that user error or a bug?

Another thing when I install some pacgakes after I got these errors:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.92-1-97356f4e229ff1d47d93048566c94202) for kmod-fs-nfs-common
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.92-1-97356f4e229ff1d47d93048566c94202) for kmod-fs-exportfs
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.92-1-97356f4e229ff1d47d93048566c94202) for kmod-crypto-des
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.92-1-97356f4e229ff1d47d93048566c94202) for kmod-crypto-hash
 etc, etc

BUT, everything is working so thank you for that very good fw, keep up the good work! 
I'm going to order the HK 1 box, what is the good dtb for that? Thank you!

ophub, 2022-01-21:
The difference between 5.15.y and 5.4.y is that there is an extra u-boot.ext. If you have tested a dtb of 5.4.y, it can be used. Generally speaking, download the firmware of 5.15.y and write it to usb, and then you only need to modify the dtb file name in /boot/uEnv.ext to use it.

xtendpayaa, 2022-01-22:
Yes of course I know that. This is why I wrote that I tested the 5.15.y kernel with the q201 dtb (what is working super good with 5.4.y) but the box not booting up, after X92 logo only colorful dots appears on the screen and thats it. With 5.4.y after X92 logo the booting operation appear in the TV I see every info. Like when I see the errors when I update the kernel from LuCI and reboot:
[image]
[image]

But maybe with the HK 1 the newer kernel going to work. Anyway I don't see nothing disability with the older kernel. So it's working nicely. Thanks your job!

ophub, 2022-01-22:
You are the first to install OpenWrt using this box. According to your feedback, there is a problem with writing emmc to this box, and it can only be used on tf card or usb for the time being.

Since the 5.4 kernel does not require the support of u-boot, it is more convenient to use. The 5.4 kernel is also a long-cycle supported series and can be used for several years.

xtendpayaa, 2022-01-22:
No, no sorry! You missunderstood something! No problem with writing emmc! The system already installed to emmc but you ask me to try the newer kernel but always the 5.4.y from emmc booting up, so I'm flashed back the original android 6 fw and now I see the booting info in the TV and can use properly q201 dtb.
When the above kernel update errors occurs I'm already in SD card not in emmc. So if I confused you sorry. 
So the conclusion is with the X92 S912 box: you can use from USB or EMMC your fw but only the 5.4.y working good, newer kernel not booting.
So everything is fine, thank you very much with your work. My HK1 box on the way...

One more question :)
/etc/opkg/distfeeds.conf I changed all links to openwrt releases/21.02.1/packages/aarch64_cortex-a53/ links and it's working.
Is this a good method to use/install latest openwrt packages or no?

ophub, 2022-01-22:
The 5.4 kernel does not require u-boot support, so it is easy to use.

However, the 5.10+ kernel requires u-boot, which will encounter many unsuitable boxes.

For software package updates, the way you replace the software source is the best, but it is not perfect, because the third-party software packages you introduce are not necessarily included in the official, that is to say, you update these third-party software packages and It is not available when dependent. Therefore, `everyone integrates it when the firmware is compiled`, rather than looking for it in the official upstream sources later, many of which cannot be found.

xtendpayaa, 2022-01-22:
Yes, thank you everything and all the informations! I just replaced the original source because you wrote me "The default is the software source server in China, you may not be able to access it. Update the software source and replace it with your domestic mirror address" because I have some signature check failed when update packages.
I think we can close this "issue", and if I have more questions when HK1 arrives from CN then I open a new.
Just I'm still curious what need to where copy for the new kernel with the HK1.
So if I understand good the u-boot-x96maxplus.bin file need to rename to u-boot.ext (TF/SD card boot file) and u-boot.emmc (EMMC boot file)?

ophub, 2022-01-23:
hk1 is a box that is already supported and does not need to be modified manually. Download the direct write emmc of s905x3 and use it.
Only a currently unknown box like your x92 requires you to replace the dtb and u-boot for testing.

Look at the description on the home page, there are some hyperlinks in the article, you can quickly find the storage directory of related files.

closed 2022-01-23 by ophub

Roxy09099, 2022-07-04:
> No, no sorry! You missunderstood something! No problem with writing emmc! The system already installed to emmc but you ask me to try the newer kernel but always the 5.4.y from emmc booting up, so [...]

@xtendpayaa can you share me the android 6 fw file you flashed for x92. i also use fw android 6 for x92 but when i try to install openwrt in emmc everything becomes error

xtendpayaa, 2022-07-04:
> @xtendpayaa can you share me the android 6 fw file you flashed for x92. i also use fw android 6 for x92 but when i try to install openwrt in emmc everything becomes error

Here it is:
https://chinagadgetsreviews.com/download-latest-android-6-0-stock-firmware-x92-tv-box-4.html

Roxy09099, 2022-07-04:
> Here it is: https://chinagadgetsreviews.com/download-latest-android-6-0-stock-firmware-x92-tv-box-4.html

the download links of this article are all dead. do you still have the firmware file?  my box is also x92 3gb ram 16gb rom

xtendpayaa, 2022-07-05:
> > > > No, no sorry! You missunderstood something! No problem with writing emmc! The system already installed to emmc but you ask me to try the newer kernel but always the 5.4.y from emmc booting up, so I'm flashed back the original android 6 fw and now I see the booting info in the TV and can use properly q201 dtb. When the above kernel update errors occurs I'm already in SD card not in emmc. So if I confused you sorry. So the conclusion is with the X92 S912 box: you can use from USB or EMMC your fw but only the 5.4.y working good, newer kernel not booting. So everything is fine, thank you very much with your work. My HK1 box on the way...
> > > > One more question :) /etc/opkg/distfeeds.conf I changed all links to openwrt releases/21.02.1/packages/aarch64_cortex-a53/ links and it's working. Is this a good method to use/install latest openwrt packages or no?
> > > 
> > > 
> > > @xtendpayaa can you share me the android 6 fw file you flashed for x92. i also use fw android 6 for x92 but when i try to install openwrt in emmc everything becomes error
> > 
> > 
> > Here it is: https://chinagadgetsreviews.com/download-latest-android-6-0-stock-firmware-x92-tv-box-4.html
> 
> @xtendpayaa the download links of this article are all dead. do you still have the firmware file? my box is also x92 3gb ram 16gb rom

Sorry. After work I'm upload for you somewhere.

Roxy09099, 2022-07-05:
> Sorry. After work I'm upload for you somewhere.

@xtendpayaa Thanks

xtendpayaa, 2022-07-05:
> @xtendpayaa Thanks

If I remember correct that was the good fw:
https://mega.nz/file/Yiw3wC6S#6DOkGgWwGiwo_5ndUMb0x1YBwZKHjlbo2ThSJ6-Cqyg