ophub/amlogic-s9xxx-openwrt #198: How can I get kmod-mt76x2u support?

Rhyno88, 2022-01-25:
I recently purchased 1xNetgear A6210 and 1xALFA AWUS036ACM USB WiFi adapters. My goal is to build a travel router using the Odroid N2+ SBC.

I've managed to get the OpenWRT build running on the Odroid N2+ but I see that the kmod-mt76x2u driver isn't installed by default.

Just looking to be pointed in the right direction. Do I have to build a new IMG that supports this driver? Or can I just install this driver using an .ipk file? Or do I have to build an .ipk file for this particular hardware?

My other challenge is that I'm unable to get internet on the device until I get the USB driver working, so need a way to install the driver without having the Odroid connected to the internet if possible.

Thanks in advance for any help :)

ophub, 2022-01-25:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/ab94a2f81c72b631da195eefd1f212589250db65

According to your needs, mt76x2u is started by default today, and you can test it after the firmware is compiled today.

closed 2022-01-25 by ophub

Rhyno88, 2022-01-25:
Amazing, thank you so much!!!