ophub/amlogic-s9xxx-openwrt #200: Can't get ip in wlan client mode

zakeru46, 2022-01-27:
i encountered problem, it seems i can connect to some AP, but in the interfaces page wwan is error network is not present. my device is hg680p, tried with internal wifi and usb wifi, no luck at all.
also, i found interesting stuff in the logs, maybe related to my problem.

```Fri Jan 28 02:01:15 2022 daemon.notice netifd: radio2 (13594): ./mac80211.sh: eval: line 1063: ipconfig: not found
Fri Jan 28 02:01:15 2022 daemon.notice netifd: radio2 (13594): ./mac80211.sh: eval: line 1064: ipconfig: not found
Fri Jan 28 02:01:15 2022 daemon.notice netifd: radio2 (13594): ./mac80211.sh: eval: line 975: ipconfig: not found
Fri Jan 28 02:01:15 2022 daemon.notice netifd: radio2 (13594): ./mac80211.sh: eval: line 1072: ipconfig: not found
Fri Jan 28 02:01:15 2022 daemon.notice netifd: radio2 (13594): ./mac80211.sh: eval: line 1073: ipconfig: not found
Fri Jan 28 02:01:16 2022 daemon.notice netifd: radio2 (13594): ./mac80211.sh: eval: line 1105: ipconfig: not found
Fri Jan 28 02:01:16 2022 daemon.notice netifd: radio2 (13594): ./mac80211.sh: eval: line 714: ipconfig: not found
Fri Jan 28 02:01:16 2022 daemon.notice netifd: radio2 (13594): ./mac80211.sh: eval: line 717: ipconfig: not found
Fri Jan 28 02:01:16 2022 daemon.notice netifd: radio2 (13594): ./mac80211.sh: eval: line 732: ipconfig: not found
Fri Jan 28 02:01:16 2022 daemon.notice netifd: radio2 (13594): ./mac80211.sh: eval: line 924: ipconfig: not found
Fri Jan 28 02:01:16 2022 daemon.notice netifd: radio2 (13594): Interface 0 setup failed: IFUP_ERROR
Fri Jan 28 02:01:16 2022 daemon.notice netifd: Wireless device 'radio2' is now up`
```

why mac80211 script calling non-existent commands in the linux??? quick look at official openwrt 21.02.1 branch, i cannot find single line calling ipconfig at all, they using iw commands if i understand correctly. thanks.

[image]

[image]

Rhyno88, 2022-01-27:
I think I'm having the same issues. Thought is was just a configuration on my end but I've tried countless things to try and get an AP or Client working. Both my USB WiFi adapters seem to be recognized and I've able to scan and add WiFi networks. Unfortunately no matter what I do I always have `Wireless is not associated` on the added network or `Network device is not present` as seen above.

ophub, 2022-01-28:
Wifi ON band 2,4ghz  choose manual channel, if auto channel, wifi off.. Wifi ON band 5ghz choose manual channel `36,40,44,48,149,153,157,165`

If your wirelees as `client mode`, `rm /etc/config/wireless  && sync && reboot` 

After that, set your wireless channel, and it should work as `server mode`

zakeru46, 2022-01-28:
> Wifi ON band 2,4ghz choose manual channel, if auto channel, wifi off.. Wifi ON band 5ghz choose manual channel `36,40,44,48,149,153,157,165`
> 
> If your wirelees as `client mode`, `rm /etc/config/wireless && sync && reboot`
> 
> After that, set your wireless channel, and it should work as `server mode`

i've got no problem when running AP mode, it's working flawlessly (with internal wifi, doesnt support 5GHz), apart from too many kernel message printed ( can we disable debug mode for that? rtl871x driver). 
The problem is when switching to client mode. i already tried that rm config wireless, no luck at all. also, i need client mode so my openwrt works as intermediate relay for cable user.

ophub, 2022-01-28:
According to the revision comments given by sibondt, I have corrected it. Please download today's repackaged firmware for testing later.

https://github.com/ophub/amlogic-s9xxx-openwrt/commit/f584229c36254161c3e385d2a19a5688c4402301

Rhyno88, 2022-01-28:
I can confirm that my client network is working now. I was able to scan and connect to the WiFi network and the device is able to associate with the network and the WAN interface. Unfortunately my second WiFi adapter is still not working is AP mode. I'll do some more testing tomorrow but just wanted to give an update regarding the Client mode for the WAN :)

zakeru46, 2022-01-28:
> I can confirm that my client network is working now. I was able to scan and connect to the WiFi network and the device is able to associate with the network and the WAN interface. Unfortunately my second WiFi adapter is still not working in AP mode. I'll do some more testing tomorrow but just wanted to give an update regarding the Client mode for the WAN :)

do you test with ophub newer release or just replacing mac80211.sh file with my link? also, support ap mode generally hit and miss with openwrt, especially for usb based adapter. I've only tested the ar9170 in the past which can do ap mode, albeit with minor issue (stalled connection with client, need to restart ap regularly)

Rhyno88, 2022-01-28:
@zakeru46 I just tested with the latest release from ophub. Well that is frustrating regarding the AP mode lol Trying to build this travel router has been a pain in the butt lol

I purchased these WiFi adapters cause they had linux kernal support and the one guy that compiled a list of WiFi adapters that worked mentioned they work in AP mode. Gonna experiment a little more but both adapters I have use the mediatek-mt7612u chipset so if the one doesn't work in AP mode I assume the other will be the same. What do people do to be able to use OpenWRT as a router then? It's great that I can connect to hotel WiFi but if I can't repeat that signal to my devices its not very helpful lol Also unfortunately the Odroid N2+ I'm using doesn't have a built in WiFi module :( 

Thanks for any help you can provide.

https://github.com/morrownr/USB-WiFi#chipset---mediatek-mt7612u---supported-in-kernel-since-linux-kernel-419-2018---mt7612u-info

zakeru46, 2022-01-29:
> @zakeru46 I just tested with the latest release from ophub. Well that is frustrating regarding the AP mode lol Trying to build this travel router has been a pain in the butt lol
> 
> I purchased these WiFi adapters cause they had linux kernal support and the one guy that compiled a list of WiFi adapters that worked mentioned they work in AP mode. Gonna experiment a little more but both adapters use the mediatek-mt7612u chipset so if the one doesn't work in AP mode I assume the other will be the same. What do people do to be able to use OpenWRT as a router then? It's great that I can connect to hotel WiFi but if I can't repeat that signal to my devices its not very helpful lol Also unfortunately the Odroid N2+ I'm using doesn't have a built in WiFi module :(
> 
> Thanks for any help you can provide.
> 
> [morrownr/USB-WiFi#chipset---mediatek-mt7612u---supported-in-kernel-since-linux-kernel-419-2018---mt7612u-info](https://github.com/morrownr/USB-WiFi#chipset---mediatek-mt7612u---supported-in-kernel-since-linux-kernel-419-2018---mt7612u-info)

yeah, i heard that mediatek, except mt7601u can do ap mode. unfortunately, finding one that have mt7612u/mt7610u is difficult here, mostly have realtek chipset (newer batch tp-link also use realtek too).  people mostly use a openwrt with a wireless router, that one have pcie wireless card/integrated wireless card. using sbc with openwrt i find bit wasteful (if you do only routing, nothing else like runnig docker, or adguard, etc). other linux distro have better compatibility because you could compile out-of-tree wireless driver and load as kernel module ( especially for realtek chipset, in my case, my rtl8192eu can do ap mode with out-of-tree driver and have great signal reception. with openwrt, you limited with rtl8xxxu which isn't great).
____
back to topic. i tested with last commit no more error radio up. seems the problem is fixed. therefore,  i will close this issue. thankyou very much.

closed 2022-01-29 by zakeru46