ophub/amlogic-s9xxx-openwrt #201: Install OpenWRT on EMMC (Odroid N2+)

Rhyno88, 2022-01-27:
Seem to be having issues installing OpenWRT on my Odroid N2+ EMMC module. I've tried both the custom dtb file and the regular N2 in the drop down list. Each time I get the below error. Currently running OpenWRT on the microSD card. I have tried to plug in the microSD card into a USB adapter but can't seem to boot from the petiboot menu for some reason. I do have a EMMC microSD adapter. Is there something I'm doing wrong or can I install the correct img directly to the EMMC module using my computer?

Thanks again for all your help.

[image]

ophub, 2022-01-28:
try booting with usb

Rhyno88, 2022-01-28:
Well that does make sense, the microSD card is mounted as mmcblk. Now as far as booting the Odroid N2+ from USB this has been a bit of a challenge.

I've tried booting the N2 using SPI into petitboot but was unable to get it to recognize the OpenWRT USB. I even updated petitboot to the latest version as I found some people have been able to boot from USB but only specific versions of Armbian.

Then I tried to load it using a microSD card modifying the `uEnv.txt` file on the boot partition and changed the `root=UUID=<etc>` to match the USB device `root=/dev/sda1` essentially using the microSD card to boot the USB drive. This did actually work but when I tried to remove the microSD card to avoid the conflict of the `mmcblk` everything failed.

Last thing I tried was flashing the OpenWRT img directly to the EMMC module and modifying the `uEnv.txt` and changing the following to `FDT=/dtb/amlogic/meson-g12b-odroid-n2-plus.dtb`. Again this failed boot fully and gave some errors.

I'm out of ideas, is there a way I can modify the EMMC module directly to get this to work? Thanks again for any help you can provide.

ophub, 2022-01-28:
Since the 5.4 kernel does not require special boot files such as u-boot, you can dd directly to emmc. Then switch to 5.10 or other kernels by updating and upgrading.

Rhyno88, 2022-01-28:
Well... that was staring me in the face in your first comment lol

Something I noticed from my earlier experiments was that the `meson-g12b-odroid-n2-plus.dtb` isn't included in the release but exists in the repo. Is this intentional?

After adding that file to the boot partition along with the other dtb files and modifying the install script to write to the `mmcblk2` device. It seems to install the files but fails to mount. 

I turned off the N2+ removed the microSD and it boots up, unfortunately not successfully and spits out some errors.

Sorry for the bad image, the only display I have seems to get cut off.

I'm trying to take advantage of the processing power on the N2+ to get maximum speeds for OpenVPN. Is that what the dtb file is for, setting the CPU voltage etc?

Thanks again for your assistance.

[image]

ophub, 2022-01-28:
Since the box is expensive, there are few users and lack of testing. If you match the appropriate dtb and u-boot files, please give your feedback and modify the default configuration file.

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/570bd948f225f9e4ff42deca7cb7291bf72ce105/amlogic-s9xxx/common-files/files/usr/sbin/openwrt-install-amlogic#L129

meson-g12b-odroid-n2.dtb
meson-g12b-odroid-n2-plus.dtb

Rhyno88, 2022-01-28:
I can confirm its working. I've added a new entry to the above install script which is just a copy of entry `45` then modified the name, dtb file and description of the cpu. After running the script in the terminal and not installing the u-boot when asked I was able to confirm that everything installed correctly and is currently running off the EMMC module.

Sorry for my lack of understanding but what is the benefit of using u-boot? Also whats the difference between `5.4` and `5.10`, I just assumed `5.10` is where most of the new development takes place and `5.4` is more stable?

I will do some more testing with the u-boot file tomorrow as I need to get some sleep before work but thanks again for all your help.

Would it help if I created a MR/PR that included some of these changes. I also found you changes for when you added the N2 to the make file for automated building of the image. If my tests work out I could potential add those changes as well. Let me know what works best for you and I'll do my best to help out ;)

ophub, 2022-01-28:
5.4 and 5.10/5.15 are the official long-term maintenance kernels of kernel.org, which will be very stable. Generally speaking, newer versions will add new feature support and be better.

However, in versions 5.10 and later, there is an additional /boot/u-boot.ext file to boot.

After you determine which dtb file is in 5.4, it has the same name in 5.10.

closed 2022-01-30 by ophub