ophub/amlogic-s9xxx-openwrt #209: brtfs-progs error while compile

Manssizz, 2022-02-15:
```
. . .
2022-02-15T20:55:43.5308935Z make[3]: *** [Makefile:100: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/btrfs-progs-v5.16.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
2022-02-15T20:55:43.5309781Z make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/btrfs-progs'
2022-02-15T20:55:43.5310318Z time: package/feeds/packages/btrfs-progs/compile#1.22#0.60#1.47
2022-02-15T20:55:43.5316816Z     ERROR: package/feeds/packages/btrfs-progs failed to build.
2022-02-15T20:55:43.5345033Z make[2]: *** [package/Makefile:116: package/feeds/packages/btrfs-progs/compile] Error 1
2022-02-15T20:55:43.5345582Z make[2]: Leaving directory '/workdir/openwrt'
2022-02-15T20:55:43.5369657Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
2022-02-15T20:55:43.5370532Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2022-02-15T20:55:43.5370911Z make[1]: Leaving directory '/workdir/openwrt'
2022-02-15T20:55:43.5495720Z ##[error]Process completed with exit code 2.
```

ophub, 2022-02-16:
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/1845365503

https://github.com/coolsnowwolf/lede

My compilation yesterday was smooth. Yesterday, the upstream source code underwent a large-scale application migration and finishing work. Please wait a few days for the adjustment and then compile it.

closed 2022-02-16 by ophub

ophub, 2022-02-16:
https://github.com/coolsnowwolf/packages/commit/faa4a71d15a793febd346cdd609698aceb54e3ba

try again

Manssizz, 2022-02-18:
another error
```
A    amlogic-s9xxx/amlogic-kernel/5.4.180/modules-5.4.180-flippy-70+o.tar.gz
Exported revision 75.
OpenWrt SoC List: [ s922x s922x-n2 s922x-reva s905x3 s905x2 s912 s912-t95z s905 s905d s905d-ki s905x s905w  ]
Kernel List: [ 5.10.101 5.4.180  ] 

(1.1) Start making OpenWrt [ s922x - 5.10.101 ]. Remaining space is 18G.
 [ s922x  -  5.10.101 ]  (1/6) extract armvirt files.
 [ s922x  -  5.10.101 ]  (2/6) extract armbian files.
cp: cannot overwrite non-directory '/home/runner/work/amlogic-s9xxx-openwrt/amlogic-s9xxx-openwrt/tmp/5.10.101/s922x/root/var' with directory '/home/runner/work/amlogic-s9xxx-openwrt/amlogic-s9xxx-openwrt/amlogic-s9xxx/common-files/files/var'
 [ s922x  -  5.10.101 ]  (3/6) refactor related files.
 [ s922x  -  5.10.101 ]  (4/6) make openwrt image.
 [ s922x  -  5.10.101 ]  (5/6) copy files to image.
 [ s922x  -  5.10.101 ]  (6/6) cleanup tmp files.
(1.1) OpenWrt made successfully. 

(1.2) Start making OpenWrt [ s922x - 5.4.180 ]. Remaining space is 17G.
 [ s922x  -  5.4.180 ]  (1/6) extract armvirt files.
 [ s922x  -  5.4.180 ]  (2/6) extract armbian files.
cp: cannot overwrite non-directory '/home/runner/work/amlogic-s9xxx-openwrt/amlogic-s9xxx-openwrt/tmp/5.4.180/s922x/root/var' with directory '/home/runner/work/amlogic-s9xxx-openwrt/amlogic-s9xxx-openwrt/amlogic-s9xxx/common-files/files/var'
Error: Process completed with exit code 1.
```

ophub, 2022-02-18:
https://github.com/ophub/amlogic-s9xxx-openwrt/runs/5242511992?check_suite_focus=true

I just tested it and it passed. You try again.

ophub, 2022-02-18:
https://github.com/Manssizz/amlogic-s9xxx-openwrt/tree/main/amlogic-s9xxx/common-files/files/var

I took a look at your error and it's because you created the wrong `var` directory. In the root directory of OpenWrt, `var points to tmp`, so you cannot create this type of directory.

Please log in the OpenWrt system, `cd / && ls -l` to check the directory structure. The directory you create in the `common-files/files` directory must be based on the directory of the OpenWrt system. For example you should create `common-files/files/www`

ophub, 2022-02-18:
Encountering problems is also a process of learning, and the more you practice, the more familiar you are.