ophub/amlogic-s9xxx-openwrt #212: WIFI?

DangerD1024, 2022-02-20:
Are there any amlogic tv box with supported internal wifi module?

I need to make wifi hotspot with 4g usb modem + small web server
Also which 4g modem is best to buy?

ophub, 2022-02-21:
Support rtl8152, rtl8153, rtl8156, etc.,
s905x3 hk1, s905x3 ugoos
The wifi of the box is not as strong as that of a professional router. It is recommended to use a professional router if the network bandwidth is above 500mb.

DangerD1024, 2022-02-21:
I've tried on ugoos am6, but there was no wifi interface...

ophub, 2022-02-21:
The wireless of Ugoos X3 is very good according to the people who use it.

DangerD1024, 2022-02-21:
I need to create small web server (PHP+MySQL) for our mobile app, so app can work without internet using wifi hotspot. Also i need to connect some 2g/3g/4g modem to sync local webserver data to main server.
I already tried some routers but they have not enough space to install php and mysql...
Maybe i should search for some router instead tv box?

ophub, 2022-02-21:
In this way, the box is a good choice.

DangerD1024, 2022-02-21:
https://www.aliexpress.com/item/4001002491240.html
This one is fine? (X3 Cube, 2+16GB)

Can you please point me on some 2g/3g/4g usb modem which works with tv box?

UMU618, 2022-02-21:
@DangerD1024 I suggest you buy a Redmi AX6 and then install openwrt, instead of TVBox.

closed 2022-02-22 by ophub