ophub/amlogic-s9xxx-openwrt #213: Unbrick Ugoos AM6 after installing

DangerD1024, 2022-02-22:
I accidentially clicked on install to nand, now box is not booting up and i unable to reflash it back using USB burning tool...
Is there any way to reflash it back to android?

DangerD1024, 2022-02-22:
everything is fine, maskrom button helped =)

ophub, 2022-02-23:
There are several ways to restore the system

How to recover if the install fails and cannot be started:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#108-how-to-recover-if-the-install-fails-and-cannot-be-started

Backup/restore the original EMMC system:
https://github.com/ophub/amlogic-s9xxx-openwrt#backuprestore-the-original-emmc-system

Which firmware are you using in Ugoos AM6, what dtb and u-boot are you using?
You can view it with the following command
`cat /etc/flippy-openwrt-release`

closed 2022-02-23 by ophub