ophub/amlogic-s9xxx-openwrt #217: How to detect CPU / SoC ?

ds2k5, 2022-03-04:
Hi,
have a Device with got only Serial Access
to u-boot.

with: **fdt print** got this INFO about CPU
```

	model = "Amlogic";
	amlogic-dt-id = "axg_s420_v03gva";
	compatible = "amlogic, axg";
	interrupt-parent = <0x00000001>;
	#address-cells = <0x00000002>;
	#size-cells = <0x00000002>;
	cpus {
		#address-cells = <0x00000002>;
		#size-cells = <0x00000000>;
		phandle = <0x00000067>;
		cpu-map {
			cluster0 {
				phandle = <0x00000068>;
				core0 {
					cpu = <0x00000002>;
				};
				core1 {
					cpu = <0x00000003>;
				};
				core2 {
					cpu = <0x00000004>;
				};
				core3 {
					cpu = <0x00000005>;
				};
			};
		};
		cpu@0 {
			device_type = "cpu";
			compatible = "arm,cortex-a53", "arm,armv8";
			reg = <0x00000000 0x00000000>;
			enable-method = "psci";
			clocks = <0x00000006 0x00000000>;
			clock-names = "cpu-cluster.0";
			cpu-idle-states = <0x00000007>;
			phandle = <0x00000002>;
		};
		cpu@1 {
			device_type = "cpu";
			compatible = "arm,cortex-a53", "arm,armv8";
			reg = <0x00000000 0x00000001>;
			enable-method = "psci";
			clocks = <0x00000006 0x00000000>;
			clock-names = "cpu-cluster.0";
			cpu-idle-states = <0x00000007>;
			phandle = <0x00000003>;
		};
		cpu@2 {
			device_type = "cpu";
			compatible = "arm,cortex-a53", "arm,armv8";
			reg = <0x00000000 0x00000002>;
			enable-method = "psci";
			clocks = <0x00000006 0x00000000>;
			clock-names = "cpu-cluster.0";
			cpu-idle-states = <0x00000007>;
			phandle = <0x00000004>;
		};
		cpu@3 {
			device_type = "cpu";
			compatible = "arm,cortex-a53", "arm,armv8";
			reg = <0x00000000 0x00000003>;
			enable-method = "psci";
			clocks = <0x00000006 0x00000000>;
			clock-names = "cpu-cluster.0";
			cpu-idle-states = <0x00000007>;
			phandle = <0x00000005>;
		};
		idle-states {
			entry-method = "arm,psci-0.2";
			cpu-sleep-0 {
				compatible = "arm,idle-state";
				arm,psci-suspend-param = <0x00010000>;
				local-timer-stop;
				entry-latency-us = <0x00001388>;
				exit-latency-us = <0x00001388>;
				min-residency-us = <0x00003a98>;
				phandle = <0x00000007>;
			};
		};
	};

At the Boot I can see this:

set vcck to 1050 mv
set vddee to 950 mv
Board ID = 3
CPU clk: 1200MHz

So CPU run with 1.2 GHZ

What SoC is this: 905 or 805 ?

Thank you !
```

ophub, 2022-03-04:
```
	- GXBB (S905) : "amlogic,meson-gxbb-"
	- GXL (S905X, S905D) : "amlogic,meson-gxl-"
	- GXM (S912) : "amlogic,meson-gxm-"
	- AXG (A113D, A113X) : "amlogic,meson-axg-"
	- G12A (S905X2, S905D2, S905Y2) : "amlogic,meson-g12a-"
```

ds2k5, 2022-03-04:
Thanks

Did I understand right I could be one of this: S905 / S095X / S912 A113D / S905X2 / S905D2 / S905Y2) ?

ophub, 2022-03-04:
Do not know

closed 2022-03-04 by ophub