ophub/amlogic-s9xxx-openwrt #225: compile kernel

kyliezero, 2022-03-09:
how to compile kernel without LOCALVERSION
example: `5.4.180-flippy+69` without `flippy+69` only `5.4.180` only

ophub, 2022-03-09:
```
# Install the necessary packages (The script has only been tested on Ubuntu-20.04-x86_64)
sudo apt-get update -y
sudo apt-get full-upgrade -y
sudo apt-get install -y $(curl -fsSL git.io/ubuntu-2004-server)

# Clone the repository to local
git clone --depth 1 https://github.com/ophub/amlogic-s9xxx-armbian.git

# Local compilation
cd amlogic-s9xxx-armbian
sudo ./recompile -d -k 5.4.180 -n -good-luck
```

closed 2022-03-09 by ophub

kyliezero, 2022-03-09:
OK I will try it

kyliezero, 2022-03-09:
> Kernel compilation and usage instructions https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/compile-kernel
> 
> ```
> # 01. Install the necessary packages (The script has only been tested on Ubuntu-20.04-x86_64)
> sudo apt-get update -y
> sudo apt-get full-upgrade -y
> sudo apt-get install -y $(curl -fsSL git.io/ubuntu-2004-server)
> 
> # 02. Clone the repository to local
> git clone --depth 1 https://github.com/ophub/amlogic-s9xxx-armbian.git
> 
> # 03. Local compilation
> cd amlogic-s9xxx-armbian
> sudo ./recompile -d -k 5.4.180 -n -good-luck
> ```

later it will be like this `5.4.180-good-luck` I want `5.4.180` without `-good-luck` :D

ophub, 2022-03-09:
cd /usr/lib/modules/
take a look at the Ubuntu and debain systems. I've never seen a system with a pure digital kernel.