ophub/amlogic-s9xxx-openwrt #228: T95Z Plus cant use openwrt or armbian firmware

Manssizz, 2022-03-19:
### Kernel 5.4
**Booting succes, installing fail**
[image]

<details>
  <summary>log openwrt-install-amlogic</summary>
  
```
root@OpenWrt:/# openwrt-install-amlogic 
AUTO_MAINLINE_UBOOT: 
SPECIFY_SOC: 
SPECIFY_DTB: 
SHARED_FSTYPE: 
Current device: T95Z Plus [ amlogic ]
/dev/mmcblk2
mmcblk2 : 31268536320 bytes
ROOTFS: mmcblk1p2
BOOT: mmcblk1p1
Please select s9xxx box model:
root@OpenWrt:/# openwrt-install-amlogic 
AUTO_MAINLINE_UBOOT: 
SPECIFY_SOC: 
SPECIFY_DTB: 
SHARED_FSTYPE: 
Current device: T95Z Plus [ amlogic ]
/dev/mmcblk2
mmcblk2 : 31268536320 bytes
ROOTFS: mmcblk1p2
BOOT: mmcblk1p1
Please select s9xxx box model:
----------------------------------------------------------------------------------------------------
1.ID  2.MODEL  3.SOC  4.FDTFILE  5.UBOOT_OVERLOAD  6.MAINLINE_UBOOT  7.ANDROID_UBOOT  8.DESCRIPTION >>>
Amlogic GXL Family >>>
 101. Phicomm N1                    s905d   4C@1512Mhz,2GB Mem,1Gb Nic,brcm43455 wifi
 102. Phicomm N1 (DMA thresh)       s905d   Same as above, when ethmac flow control is off
 103. hg680p & b860h                s905x   4C@1512Mhz,100M Nic
 104. X96-Mini & TX3-Mini           s905w   4C@1512Mhz,100M Nic
 105. MECOOL KI Pro                 s905d   2G/16G,1Gb Nic
Amlogic GXM Family >>>
 201. Octopus Planet                s912    4C@1512Mhz+4C@1000Mhz,2GB Mem,1Gb Nic
 202. H96 Pro Plus                  s912    2G/32G,1Gb Nic
 203. Tanix-TX92                    s912    3GB DDR4 32GB eMMC,1.5GHz,5G WIFI,1Gb Nic
 204. VORKE-Z6-Plus                 s912    3GB DDR3 32GB eMMC5.0,1.5Ghz,TF CARD Support 1~32GB,1Gb Nic
 205. T95Z Plus                     s912    3G+32G,Octa-Core,2.4/5.8G Dual-Band Wi-Fi/1Gb LAN 64Bit BT4.0
 206. MECOOL M8S Pro L              s912    2G RAM,3G RAM,16G ROM,32G ROM,100M Nic
Amlogic G12A Family >>>
 301. X96 Max 4GB                   s905x2  4C@1908Mhz,4GB Mem,1Gb Nic
 302. X96 Max 2GB                   s905x2  4C@1908Mhz,2GB Mem,100M Nic
 303. MECOOL KM3 4G                 s905x2  4C@1908Mhz,4+64G/128G,2.4G/5G WiFi,Bluetooth 4.1,100M Nic
 304. E900V22C/D                    s905l3a 4C@1908Mhz,S905L3A+B,2+8G,USB2.0x2,LAN 100M Nic,uwe5621ds wifi(no work)
Amlogic G12B Family >>>
 401. Beelink GT-King               s922x   2C@1800Mhz(A53)+4C@1908Mhz(A73),4GB Mem,1Gb Nic,brcm4356 wifi
 402. Beelink GT-King Pro           s922x   2C@1800Mhz(A53)+4C@1908Mhz(A73),4GB Mem,1Gb Nic,brcm4356 wifi
 403. Beelink GT-King Pro H         s922x   S922X-H,2C@1800Mhz(A53)+4C@2208Mhz(A73),4GB Mem,1Gb Nic,brcm4356 wifi
 404. Beelink GT-King Pro Rev_A     s922x   2C@1800Mhz(A53)+4C@1908Mhz(A73),4GB Mem,1Gb Nic,brcm4356 wifi
 405. Hardkernel ODroid N2          s922x   2C@1800Mhz(A53)+4C@1908Mhz(A73),4GB Mem,1Gb Nic
 406. UGOOS AM6 Plus                s922x   2C@1800Mhz(A53)+4C@1908Mhz(A73),4GB Mem,1Gb Nic,brcm4398 wifi
 407. Khadas VIM3                   a311d   4C@2.2Ghz+2C@1.8Ghz,PCIe+USB 3.0,1Gb Nic,brcm4398 wifi
Amlogic SM1 Family >>>
 501. X96 Max+                      s905x3  4C@2100Mhz,4GB Mem,1Gb Nic,rtl8822cs wifi
 502. X96 Max+ (OverClock)          s905x3  4C@2208Mhz,4GB Mem,1Gb Nic,rtl8822cs wifi
 503. HK1 Box                       s905x3  4C@2100Mhz,4GB Mem,1Gb Nic,brcm4339 wifi
 504. HK1 Box (OverClock)           s905x3  4C@2208Mhz,4GB Mem,1Gb Nic,brcm4339 wifi
 505. H96 Max X3                    s905x3  4C@2100Mhz,4GB Mem,1Gb Nic,brcm4339 wifi
 506. H96 Max X3 (OverClock)        s905x3  4C@2208Mhz,4GB Mem,1Gb Nic,brcm4339 wifi
 507. Ugoos X3                      s905x3  4C@2100Mhz,2(Cube)/4(Pro,Plus)GB Mem,1Gb Nic,brcm43455/43456 wifi
 508. Ugoos X3 (OverClock)          s905x3  4C@2208Mhz,2(Cube)/4(Pro,Plus)GB Mem,1Gb Nic,brcm43455/43456 wifi
 509. TX3 1Gb                       s905x3  4C@2100Mhz,4GB Mem,1Gb Nic,bcm4330 wifi
 510. TX3 1Gb (OverClock)           s905x3  4C@2208Mhz,4GB Mem,1Gb Nic,bcm4330 wifi
 511. TX3 100Mb                     s905x3  4C@2100Mhz,4GB Mem,100Mb Nic,bcm4330 wifi
 512. TX3 100Mb (OverClock)         s905x3  4C@2208Mhz,4GB Mem,100Mb Nic,bcm4330 wifi
 513. X96 Air 1Gb                   s905x3  4C@2100Mhz,2GB Mem,1Gb Nic,Wifi
 514. X96 Air 100Mb                 s905x3  4C@2100Mhz,2GB Mem,100M Nic,Wifi
 515. A95XF3 Air 1Gb                s905x3  4C@2100Mhz,2GB Mem,1Gb Nic,Wifi
 516. A95XF3 Air 100Mb              s905x3  4C@2100Mhz,2GB Mem,100M Nic,Wifi
Other >>>
   0. Unknown                       Other   Enter the dtb file name of your box
----------------------------------------------------------------------------------------------------
Please choose: 205
FDT Value [ meson-gxm-t95z-plus.dtb ]
This Kernel [ 5.4.185-flippy-71+o ]
Backup bootloader -> [ BackupOldBootloader.img ] ... 
[  232.840770] print_req_error: 14 callbacks suppressed
[  232.840785] blk_update_request: I/O error, dev mmcblk2, sector 0 op 0x0:(READ) flags 0x80700 phys_seg 63 prio class 0
[  232.851967] blk_update_request: I/O error, dev mmcblk2, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  232.860812] buffer_io_error: 2 callbacks suppressed
[  232.860821] Buffer I/O error on dev mmcblk2, logical block 0, async page read
dd: /dev/mmcblk2: I/O error
Backup bootloader complete.

A total of [ 0 ] old partitions on EMMC will be deleted

Welcome to fdisk (util-linux 2.37.3).
Changes will remain in memory only, until you de[  232.983123] blk_update_request: I/O error, dev mmcblk2, sector 0 op 0x0:(READ) flags 0x80700 phys_seg 0
[  232.994720] blk_update_request: I/O error, dev mmcblk2, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  233.002128] Buffer I/O error on dev mmcblk2, logical block 0, async page read
[  233.011759] blk_update_request: I/O error, dev mmcblk2, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  233.019253] Buffer I/O error on dev mmcblk2, logical block 0, async page read
cide to write them.
Be careful before using the write command.

[  233.037843] blk_update_request: I/O error, dev mmcblk2, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0

BusyBox v1.35.0 (2022-03-16 05:37:25 UTC) multi-call binary.

Usage: dd [if=FILE] [of=FILE] [ibs=N obs=N/bs=N] [count=N] [skip=N] [seek=N]
        [conv=notrunc|noerror|sync|fsync]
        [iflag=skip_bytes|count_bytes|fullblock|direct] [oflag=seek_bytes|append|direct]

Copy a file with converting and formatting

        if=FILE         Read from FILE instead of stdin
        of=FILE         Write to FILE instead of stdout
        bs=N            Read and write N bytes at a time
        ibs=N           Read N bytes at a time
        obs=N           Write N bytes at a time
        count=N         Copy only N input blocks
        skip=N          Skip N input blocks
        seek=N          Skip N output blocks
        conv=notrunc    Don't truncate output file
        conv=noerror    Continue after read errors
        conv=sync       Pad blocks with zeros
        conv=fsync      Physically write data out before finishing
        conv=swab       Swap every pair of bytes
        iflag=skip_bytes        skip=N is in bytes
        iflag=count_bytes       count=N is in bytes
        oflag=seek_bytes        seek=N is in bytes
        iflag=direct    O_DIRECT input
        oflag=direct    O_DIRECT output
        iflag=fullblock Read full blocks
        oflag=append    Open output in append mode

N may be suffixed by c (1), w (2), b (512), kB (1000), k (1024), MB, M, GB, G
dd: error writing '/dev/mmcblk2': I/O error
1+0 records in
0+0 records out

```
   
</details>

**i/o error on emmc**
[image]

### Kernel 5.15
**Can't booting**
[image]

_PS: My stb works well on Android OS._

ophub, 2022-03-19:
Did you download this firmware from my repository? The file mount information is incorrect. You can download the latest firmware from me and test it.

Manssizz, 2022-03-19:
HATUR NUHUN GAN

ophub, 2022-03-19:
You can test it on the USB first, you can't write to EMMC, because the `UUID` of the extlinux/extlinux.conf file has not been modified in the installation script, I will improve it later

Manssizz, 2022-03-19:
i'll try after latest update. seems you still update some file

ophub, 2022-03-19:
The `armbian` system has been adjusted, you first download the latest firmware and try

ophub, 2022-03-20:
I also patched the install and update scripts of the t95z-plus, try the latest firmware again.
In addition, you can test whether it can be started using uEnv.txt

https://github.com/ophub/amlogic-s9xxx-openwrt/commit/eb5ae1e03cc20c28aae4f014ed052e6a66e01572

Manssizz, 2022-03-20:
thank you. i will try and report soon

ophub, 2022-03-20:
Me and my friends and your feedback

Manssizz, 2022-03-20:
booting succesfuly but have a mistake request partition block device. so.. they still detect i/o errors *cmiw
emmc boot partition = `mmcblk2boot0`
sdcard boot parititon = `none`

**Partition table**
```
root@OpenWrt:~# lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
mmcblk1      179:0    0 29.7G  0 disk 
├─mmcblk1p1  179:1    0  243M  0 part /boot
└─mmcblk1p2  179:2    0  975M  0 part /opt/docker/btrfs
                                      /opt/docker
                                      /
mmcblk2      179:32   0 29.1G  0 disk 
mmcblk2boot0 179:64   0    4M  1 disk 
mmcblk2boot1 179:96   0    4M  1 disk 
```
**Boot log**
```
...
[   25.301196] usbserial: USB Serial support registered for GSM modem (1-port)
[   25.313723] kmodloader: done loading kernel modules from /etc/modules.d/*
[   27.893456] print_req_error: 224 callbacks suppressed
[   27.893478] blk_update_request: I/O error, dev mmcblk2, sector 0 op 0x0:(READ) flags 0x80700 phys_seg 4 prio class 0
[   27.907240] blk_update_request: I/O error, dev mmcblk2, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   27.916315] buffer_io_error: 107 callbacks suppressed
[   27.916322] Buffer I/O error on dev mmcblk2, logical block 0, async page read
[   27.931618] blk_update_request: I/O error, dev mmcblk2, sector 61071232 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   27.944476] blk_update_request: I/O error, dev mmcblk2, sector 61071232 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   27.957767] Buffer I/O error on dev mmcblk2, logical block 7633904, async page read
[   27.968167] blk_update_request: I/O error, dev mmcblk2boot0, sector 0 op 0x0:(READ) flags 0x80700 phys_seg 4 prio class 0
[   27.984416] blk_update_request: I/O error, dev mmcblk2boot0, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   27.998815] Buffer I/O error on dev mmcblk2boot0, logical block 0, async page read
[   28.010798] blk_update_request: I/O error, dev mmcblk2boot0, sector 8064 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   28.024809] blk_update_request: I/O error, dev mmcblk2boot0, sector 8064 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   28.040481] Buffer I/O error on dev mmcblk2boot0, logical block 1008, async page read
[   28.050550] blk_update_request: I/O error, dev mmcblk2boot1, sector 0 op 0x0:(READ) flags 0x80700 phys_seg 4 prio class 0
...
```
**Installing fail**
[image]

ophub, 2022-03-20:
try armbian

ophub, 2022-03-20:
Since the install and update scripts use unifreq files, I added support for extlinux.conf by patching the original install and update scripts today, but I don't like adding extra patches for this one device, so I've removed the T95 -Plus devices support OpenWrt, Armbian continues to retain, if you test armbian can be used, you can refer to the practice to modify the OpenWrt settings.

closed 2022-03-20 by ophub