ophub/amlogic-s9xxx-openwrt #231: T95z plus can not initialize PHY ethernet

Richard-Chueng, 2022-03-21:
Hi,
1) No problem to boot the usb with latest firmware (openwrt_s912_k5.15.30_2022.03.20.2154.img.gz), but after install to emmc, it shows success, but can not initialize the ethernet after reboot.
2) I also try to reinstall the android firmware before install openwrt, but the problem is the same
[image]
.

ophub, 2022-03-21:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/b099d1a8a4f313b799ebb8eadeafcd569b0c67a2
I have removed support for T95 devices, the current firmware is not suitable for T95 use

ophub, 2022-03-21:
https://github.com/ophub/amlogic-s9xxx-armbian/releases

Find the firmware named `Armbian_22.05.0_Aml_s912-t95z_focal_5.15.30_xxx.img.gz` here, and see if Armbian can be used normally on T95. If so, OpenWrt can be modified with the above method.

closed 2022-03-21 by ophub

Richard-Chueng, 2022-03-21:
Hi,

I did try to install armbian firmware named Armbian_22.05.0_Aml_s912-t95z_focal_5.15.30_xxx.img.gz as you suggest above,
1) when booting with USB, it fails to raise the network interfaces;
[image]

Richard-Chueng, 2022-03-21:
2) after some minutes, it can boot from usb, and also can not boot from emmc after successful installation.
[image]

ophub, 2022-03-21:
You plug in the network cable, without the network cable, of course there is no network

ophub, 2022-03-21:
Thank you for your feedback. It seems that this dtb is not applicable. I deleted T95 from the supported list and added it later when I have his compiled source code.

Richard-Chueng, 2022-03-21:
Thanks for your reply, but i have writed image openwrt_s912-t95z_k5.15.25_2022.02.26.2120.img.gz to usb disk, and boot from usb without any problem for almost a month.
PS: only booting from usb will work, always fails booting from EMMC.

ophub, 2022-03-21:
The `meson-gxm-t95z-plus.dtb` of this firmware has no compiled source code. It is a fixed file shared by netizens. It may not be used in different versions of the kernel such as 5.4/5.10. After writing emmc, the network card cannot be driven. I removed it first and added it later when I found the dts.

puccio823, 2022-11-18:
hi last night i did some tests and finally it booted with **meson-gxbb-p200.dtb** file
[image]
but now I have this problem
[image]