ophub/amlogic-s9xxx-openwrt #235: emmc u-boot question...

rossbcan, 2022-04-01:
Hi;

Box: Tanix TX9 Pro
u-boot: u-boot-s905x-s912.bin
FDT: /dtb/amlogic/meson-gxm-q201.dtb
same settings as "MECOOL M8S Pro L"

boots properly from mSata disk in USB adapter or USB flash disk. Fails boot from SD (not an issue for me)

Several questions:

At "https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-docs/config_correspondence_of_amlogic_s9xxx_tv_box.md" instructions.

Does "NA" entry for "MAINLINE_UBOOT" for box models mean there is no mainline u-boot (for emmc) available?

Does NA mean the external media u-boot is identical to u-boot for emmc?

If NA means no emmc u-boot, what are my options for downloading / compiling an emmc version of mainline u-boot equivalent to u-boot-s905x-s912.bin?

Is it worth trying  Khadas VIM2 (mainline) emmc u-boot, same architecture?

Thanks;
Bill

ophub, 2022-04-02:
https://github.com/ophub/luci-app-amlogic/blob/main/luci-app-amlogic/root/usr/sbin/openwrt-install-amlogic#L92-L120

```
# box model database
# The field separator is :
# " " or "" or NA or NULL means this field is null
# The fields list:
# 1.  id
# 2.  model name
# 3.  SOC
# 4.  FDTFILE
# 5.  UBOOT_OVERLOAD
# 6.  MAINLINE_UBOOT
# 7.  ANDROID_UBOOT
# 8.  brief description
#
# Amlogic GXM Family
21:Octopus Planet:s912:meson-gxm-octopus-planet.dtb:u-boot-zyxq.bin:NA:NA:4C@1512Mhz+4C@1000Mhz,2GB Mem,1Gb Nic

```

As mentioned in the remarks, NA does not have this file.

`There are two types of U-BOOT`, one of which is `MAINLINE_UBOOT`, `which will use dd to write to the emmc partition`, replacing the u-boot of the original Android system. The other one is `UBOOT_OVERLOAD`, which is `renamed to u-boot.ext and u-boot.emmc` in the boot partition for use, and boots higher `kernel versions after 5.10`.

If you use the firmware of `5.4 kernel`, the above two kinds of U-BOOT are `not required` (if there is, it can also be added).

If you use firmware with `kernel version 5.10 and above`, `you must have UBOOT_OVERLOAD`, which is responsible for reloading and booting. If it is missing, it will definitely fail to start. But MAINLINE_UBOOT can be without (generally it can be started without it, if some will be stable)

closed 2022-04-02 by ophub

rossbcan, 2022-04-02:
Forgot to mention: your amlogic work is awesome, well done, very much appreciated

It is still unclear to me whether $UBOOT_OVERLOAD is used (flashed to emmc) as $MAINLINE_UBOOT for my model.

I have inspected the update scripts and configuration files to understand how install to emmc works.
I decided to take a risk of bricking box by using luci install. Success.

Procedure:

Edit "/etc/flippy-openwrt-release" for correct config:
`PLATFORM='amlogic'
KERNEL_VERSION='5.15.31'
K510='1'
FDTFILE='meson-gxm-q201.dtb'
UBOOT_OVERLOAD='u-boot-s905x-s912.bin'
MAINLINE_UBOOT='/lib/u-boot'
ANDROID_UBOOT='/lib/u-boot'
SOC='s912'
`

(Luci) Select "MECOOL M8S Pro L" for emmc install.

Went through install procedure, successfully booted from emmc.

What confused me is that according to "/usr/sbin/openwrt-update-amlogic" (excerpt):
`if [[ "${FLASH_MAINLINE_UBOOT}" -eq "1" ]]; then
    echo -e "Write Mainline bootloader: [ ${MAINLINE_UBOOT} ]"
    dd if=${P2}${MAINLINE_UBOOT} of=/dev/${EMMC_NAME} bs=1 count=444 conv=fsync
    dd if=${P2}${MAINLINE_UBOOT} of=/dev/${EMMC_NAME} bs=512 skip=1 seek=1 conv=fsync
elif [[ -n "${ANDROID_UBOOT}" && -f "${P2}${ANDROID_UBOOT}" ]]; then
    echo -e "Write Android bootloader: [ ${ANDROID_UBOOT} ]"
    dd if=${P2}${ANDROID_UBOOT} of=/dev/${EMMC_NAME} bs=1 count=444 conv=fsync
    dd if=${P2}${ANDROID_UBOOT} of=/dev/${EMMC_NAME} bs=512 skip=1 seek=1 conv=fsync
else
    echo "Did not change the original bootloader."
fi
`

is that ${P2}${MAINLINE_UBOOT} == /root/lib/u-boot is empty, so dd fails and existing uboot in flash is not touched.

So, it appears that boxes without $MAINLINE_UBOOT use existing (original) flash u-boot and chainload to (emmc) /boot/u-boot.emmc which equals $UBOOT_OVERLOAD file in /etc/flippy-openwrt-release

I note that once installed to emmc, always boots from emmc and does not boot from USB if a USB boot disk is installed. Can a boot device priority be set?

Please confirm or correct my conclusions.

Thanks;
Bill

ophub, 2022-04-02:
The operation of writing emmc from usb has nothing to do with /etc/flippy-openwrt-release.
To add your device configuration in `/use/sbin/openwrt-install-amlogic`, copy a line, customize a serial number, modify the configuration, and select the serial number you customized during installation