ophub/amlogic-s9xxx-openwrt #236: SIGSEGV: kernel/time/hrtimer.c:1047 hrtimer_forward+0x80/0xb0

rossbcan, 2022-04-02:
I'm not sure where to report this:

`root@OpenWrt:/# [  429.403786] ------------[ cut here ]------------
[  429.410383] WARNING: CPU: 2 PID: 23 at kernel/time/hrtimer.c:1047 hrtimer_forward+0x80/0xb0
[  429.417616] Modules linked in: pppoe ppp_async option cdc_mbim ath9k_htc ath9k_common usb_wwan sr9700 smsc95xx sierra_net rtl8192cu rtl8192c_common rtl_                                                                                                                                                                  usb rt2800usb rt2800lib rt2500usb rndis_host qmi_wwan pppox ppp_generic plusb pl2303 mt76x0u mt76x0_common mcs7830 kalmia iptable_nat ipt_REJECT huawei_cdc                                                                                                                                                                  _ncm dm9601 cp210x ch341 cdc_subset cdc_ncm cdc_ether cdc_eem carl9170 brcmfmac br_netfilter ax88179_178a ath9k_hw asix xt_time xt_tcpudp xt_tcpmss xt_stat                                                                                                                                                                  istic xt_state xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_                                                                                                                                                                  connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY vhci_hc                                                                                                                                                                  d usbserial usbnet usblp usbip_host usbip_core ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_al                                                                                                                                                                  auda slhc rtlwifi rtl8xxxu rtl8188fu rtl8187 rtl8150 rt73usb rt2x00usb
[  429.417943]  rt2x00lib
[  429.430359] cpufreq: __target_index: Failed to change cpu frequency: -5
[  429.521324]  rsi_usb rsi_91x r8723bs(C) r8712u(C) r8188eu(C) r8152 pegasus panfrost nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nf                                                                                                                                                                  t_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nf_tables nf                                                                                                                                                                  _reject_ipv4 nf_nat nf_flow_table nf_dup_netdev nf_conntrack_netlink nf_conncount mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_com                                                                                                                                                                  mon mt7615_common mt7601u mt76_usb mt76_connac_lib mt76 meson_gxbb_wdt macvlan mac80211 kaweth iptable_raw iptable_mangle iptable_filter ipt_ECN ipheth ip_                                                                                                                                                                  tables hso gpu_sched btrsi bridge brcmutil bluetooth ath6kl_usb ath6kl_core ath 88x2cs 8189fs cfg80211 rfkill xt_set ip_set_list_set ip_set_hash_netportnet                                                                                                                                                                   ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport                                                                                                                                                                   ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac
[  429.521571]  ip_set_bitmap_ip ip_set nfnetlink nf_log_syslog ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3                                                                                                                                                                   nfsd nfs bonding dummy ipip rpcsec_gss_krb5 auth_rpcgss veth tun lockd sunrpc grace hfsplus hfs cdrom cifs cifs_md4 cifs_arc4 dns_resolver dm_mirror dm_reg                                                                                                                                                                  ion_hash dm_log dm_crypt dm_mod eeprom_93cx6 xhci_pci xhci_pci_renesas ohci_pci uhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_fsl squashfs reiserfs
[  429.521718] CPU: 2 PID: 23 Comm: kworker/2:0 Tainted: G         C        5.15.31-flippy-71+ #92
[  429.521727] Hardware name: Amlogic Meson GXM (S912) Q201 Development Board (DT)
[  429.521738] Workqueue: events_freezable_power_ thermal_zone_device_check
[  429.521770] pstate: 200000c5 (nzCv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  429.521780] pc : hrtimer_forward+0x80/0xb0
[  429.521800] lr : txdone_hrtimer+0x118/0x140
[  429.521812] sp : ffffffc00999be00
[  429.521815] x29: ffffffc00999be00 x28: ffffff80bf740fe8 x27: 00000063fad534a3
[  429.521828] x26: ffffff80bf740fd8 x25: ffffff80023ad0d0 x24: 0000000000000003
[  429.521839] x23: 00000000000002e8 x22: ffffff80023ad3a8 x21: ffffff8001a4e9e0
[  429.521849] x20: 00000000000f4240 x19: ffffff80023ad3e8 x18: 0000000000000000
[  429.521860] x17: ffffffc0b6152000 x16: ffffffc00999c000 x15: 0000000000004000
[  429.521870] x14: 0000000000000006 x13: 000000000003ed6c x12: 0000000029ab3c6b
[  429.521881] x11: 0039756e92419792 x10: 0000000000000001 x9 : 00000063fad539d8
[  429.521891] x8 : 0000000000000000 x7 : 0000000000000000 x6 : 0000000000000000
[  429.521901] x5 : 0000000000000000 x4 : 0000000000000008 x3 : 000000000000000f
[  429.521911] x2 : 00000000000f4240 x1 : 00000063fad542cc x0 : ffffff80023ad3e8
[  429.521923] Call trace:
[  429.521928]  hrtimer_forward+0x80/0xb0
[  429.521935]  __hrtimer_run_queues+0x124/0x1d0
[  429.521943]  hrtimer_interrupt+0xec/0x3ac
[  429.521951]  arch_timer_handler_phys+0x3c/0x4c
[  429.521963]  handle_percpu_devid_irq+0x98/0x17c
[  429.521975]  handle_domain_irq+0x80/0xe4
[  429.521990]  gic_handle_irq+0x68/0xfc
[  429.522004]  call_on_irq_stack+0x2c/0x38
[  429.522016]  do_interrupt_handler+0x3c/0x50
[  429.522027]  el1_interrupt+0x30/0x5c
[  429.522043]  el1h_64_irq_handler+0x18/0x24
[  429.522051]  el1h_64_irq+0x78/0x7c
[  429.522057]  _raw_spin_unlock_irq+0x18/0x58
[  429.522068]  wait_for_completion_timeout+0x14/0x20
[  429.522075]  mbox_send_message+0xe0/0x158
[  429.522081]  scpi_send_message+0x19c/0x230
[  429.522094]  scpi_sensor_get_value+0x40/0x90
[  429.522102]  scpi_read_temp+0x4c/0xbc
[  429.522109]  of_thermal_get_temp+0x28/0x3c
[  429.522117]  thermal_zone_get_temp+0x58/0x84
[  429.522124]  thermal_zone_device_update+0xc0/0x2cc
[  429.522131]  thermal_zone_device_check+0x18/0x24
[  429.522138]  process_one_work+0x1d4/0x2bc
[  429.522150]  worker_thread+0x278/0x514
[  429.522156]  kthread+0x144/0x170
[  429.522164]  ret_from_fork+0x10/0x20
[  429.522171] ---[ end trace 90c72eb3bb1f0478 ]---
`

ophub, 2022-04-02:
What equipment, soc model, network card chip, what kernel is used

rossbcan, 2022-04-03:
Hi ophub;

The problem: Periodic SIGSEGV from early boot

dmesg | grep kernel/time/hrtimer.c
`[  429.410383] WARNING: CPU: 2 PID: 23 at kernel/time/hrtimer.c:1047 hrtimer_forward+0x80/0xb0
[ 1434.707159] WARNING: CPU: 2 PID: 23 at kernel/time/hrtimer.c:1047 hrtimer_forward+0x80/0xb0
[16036.919599] WARNING: CPU: 2 PID: 23 at kernel/time/hrtimer.c:1047 hrtimer_forward+0x80/0xb0
[63397.474475] WARNING: CPU: 2 PID: 23 at kernel/time/hrtimer.c:1047 hrtimer_forward+0x80/0xb0
[66492.207278] WARNING: CPU: 2 PID: 23 at kernel/time/hrtimer.c:1047 hrtimer_forward+0x80/0xb0`

Does not seen to break anything, recovers.

Box: Tanix TX9 Pro
u-boot: u-boot-s905x-s912.bin
FDT: /dtb/amlogic/meson-gxm-q201.dtb
same settings as "MECOOL M8S Pro L"
All build / configs (apart from u-boot, FDT selections) are stock, unmodified. No extra packages installed.

uname -a: Linux OpenWrt 5.15.31-flippy-71+ #92 SMP PREEMPT Wed Mar 23 20:02:04 CST 2022 aarch64 GNU/Linux
Machine model: Amlogic Meson GXM (S912) Q201 Development Board

dmesg: grep eth0:
`[   26.314752] meson8b-dwmac c9410000.ethernet eth0: PHY [0.e40908ff:08] driver [Meson GXL Internal PHY] (irq=44)
[   26.329789] meson8b-dwmac c9410000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   26.351821] meson8b-dwmac c9410000.ethernet eth0: No Safety Features support found
[   26.358952] meson8b-dwmac c9410000.ethernet eth0: PTP not supported by HW
[   26.367648] meson8b-dwmac c9410000.ethernet eth0: configuring for phy/rmii link mode
[   26.378721] br-lan: port 1(eth0) entered blocking state
[   26.383904] br-lan: port 1(eth0) entered disabled state
[   26.389188] device eth0 entered promiscuous mode
[   28.155560] meson8b-dwmac c9410000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   28.163336] br-lan: port 1(eth0) entered blocking state
[   28.168191] br-lan: port 1(eth0) entered forwarding state`

dmesg | grep eth1: (USB eth adapter)
'[   20.100289] r8152 1-2.4:1.0 eth1: v1.12.12
[   29.764150] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   29.769487] r8152 1-2.4:1.0 eth1: carrier on'

It may be related to this patch required: https://lkml.iu.edu/hypermail/linux/kernel/2010.1/07482.html

possible patch:
`--- a/drivers/mailbox/mailbox.c 2021-12-03 06:46:58.258249938 -0500
+++ b/drivers/mailbox/mailbox.c 2021-12-03 06:58:08.386555305 -0500
@@ -82,9 +82,12 @@
 exit:
      	spin_unlock_irqrestore(&chan->lock, flags);

-	if (!err && (chan->txdone_method & TXDONE_BY_POLL))
-               /* kick start the timer immediately to avoid delays */
-               hrtimer_start(&chan->mbox->poll_hrt, 0, HRTIMER_MODE_REL);
+	/* kick start the timer immediately to avoid delays */
+	if (!err && (chan->txdone_method & TXDONE_BY_POLL)) {
+               /* but only if not already active */
+               if (!hrtimer_active(&chan->mbox->poll_hrt))
+                       hrtimer_start(&chan->mbox->poll_hrt, 0, HRTIMER_MODE_REL);
+	}
 }

 static void tx_tick(struct mbox_chan *chan, int r)
@@ -122,11 +125,10 @@
                struct mbox_chan *chan = &mbox->chans[i];

                if (chan->active_req && chan->cl) {
+                       resched = true;
                        txdone = chan->mbox->ops->last_tx_done(chan);
                        if (txdone)
                                tx_tick(chan, 0);
-                       else
-                               resched = true;
                }
        }
`

Regards;
Bill

ophub, 2022-04-03:
I don't understand what you're saying, what's your problem?

rossbcan, 2022-04-03:
The problem is periodic SIGSEGV which appears to be some sort of race condition causing segmentation faults in unmodified (stock) amlogic-s9xxx-openwrt distribution. I am surprised this has not been reported before. I suspect the patch above is required. I have done previous amlogic porting (not your distribution), experienced this issue. Patch referenced above solved it.

Regards;
Bill

ophub, 2022-04-03:
Thank you for sharing. I asked teacher unifreq to check it, the current 4 kernels have already been patched.

closed 2022-04-03 by ophub

rossbcan, 2022-04-03:
Patch is done. I will re-open if further information or solution found.

...B

ophub, 2022-04-03:
ok