ophub/amlogic-s9xxx-openwrt #237: Question: howto re-create openwrt-armvirt-64-default-rootfs.tar.gz

rossbcan, 2022-04-04:
Hi;

For reasons tedious to explain I need to recreate / compile from source openwrt-armvirt-64-default-rootfs.tar.gz to add my own custom packages and use glibc instead of musl. The changes are extensive and easier than using your patch methodology.

git clone openwrt
git checkout v21.02.02
used .config from https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config/openwrt-21.02

compiled successfully, but many packages missing.

Do you have the full .config for creating openwrt-armvirt-64-default-rootfs.tar.gz?

need these two files:
openwrt/.config
openwrt/target/linux/armvirt/64/config-5.10

Can you help, or point me to who built / repository openwrt-armvirt-64-default-rootfs.tar.gz

FYI: I am porting my X86 project to amlogic: www.rossco.org

Thanks;
Bill

ophub, 2022-04-05:
https://github.com/breakings/OpenWrt

He has, you can go to his warehouse to check and consult

closed 2022-04-05 by ophub

predators46, 2022-04-06:
@rossbcan 

do you want to use glibc?