ophub/amlogic-s9xxx-openwrt #239: hci_uart_bcm serial0-0: serial0-0 supply vbat not found, using dummy regulator

0neday, 2022-04-07:
```
[   11.302973] Bluetooth: HCI UART driver ver 2.3
[   11.303467] Bluetooth: HCI UART protocol H4 registered
[   11.308623] Bluetooth: HCI UART protocol BCSP registered
[   11.313898] Bluetooth: HCI UART protocol LL registered
[   11.318916] Bluetooth: HCI UART protocol ATH3K registered
[   11.324295] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   11.330677] Bluetooth: HCI UART protocol Intel registered
[   11.336032] Bluetooth: HCI UART protocol Broadcom registered
[   11.336206] hci_uart_bcm serial0-0: serial0-0 supply vbat not found, using dummy regulator
[   11.341477] Bluetooth: HCI UART protocol QCA registered
[   11.349732] hci_uart_bcm serial0-0: serial0-0 supply vddio not found, using dummy regulator
[   11.354788] Bluetooth: HCI UART protocol AG6XX registered
[   11.354831] Bluetooth: HCI UART protocol Marvell registered
[   11.607904] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   11.608296] Bluetooth: HIDP socket layer initialized
[   11.619412] PPP generic driver version 2.4.2
[   11.620470] NET: Registered protocol family 24
[   11.625961] Bluetooth: RFCOMM TTY layer initialized
[   11.627321] Bluetooth: RFCOMM socket layer initialized
[   11.632414] Bluetooth: RFCOMM ver 1.11
[   11.637709] usbcore: registered new interface driver rndis_host
[   11.644567] Bridge firewalling registered
[   11.651881] kmodloader: done loading kernel modules from /etc/modules.d/*
[   11.714265] Bluetooth: hci0: BCM: chip id 107
[   11.714568] Bluetooth: hci0: BCM: features 0x2f
[   11.718984] Bluetooth: hci0: BCM4345C0
[   11.721220] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[   12.674345] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0230
[   16.665027] meson8b-dwmac c9410000.ethernet eth0: PHY [0.2009087f:00] driver [RTL8211F Gigabit Ethernet]
[   16.730286] meson8b-dwmac c9410000.ethernet eth0: No Safety Features support found
[   16.732233] meson8b-dwmac c9410000.ethernet eth0: PTP not supported by HW
[   16.739003] meson8b-dwmac c9410000.ethernet eth0: configuring for phy/rgmii link mode
[   19.434936] meson8b-dwmac c9410000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   19.437761] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   22.442654] EXT4-fs (mmcblk2p4): recovery complete
[   22.443186] EXT4-fs (mmcblk2p4): mounted filesystem with ordered data mode. Opts: (null)
[   30.573999] wireguard: WireGuard 1.0.20211208 loaded. See www.wireguard.com for information.
[   30.576812] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
```
bluetooth not work

ophub, 2022-04-07:
Which kernel and what are you doing with bluetooth in openwrt? which plugin?

0neday, 2022-04-07:
5.4.170
use bluetooth to get ble sensor's data by homeasssisstant, such as xiaomi temperature meter.
homeassisstant running on docker, no luci-app plugin.
I install bluez-utils by opkg

ophub, 2022-04-08:
https://github.com/ophub/kernel/tree/main/pub/eol/5.16.18
Change the kernel test of 5.16.18

closed 2022-04-10 by ophub

0neday, 2022-04-10:
nop, I have fixed 

```
root@OpenWRT:~# hciconfig
hci0:   Type: Primary  Bus: UART
        BD Address: 43:45:C0:00:1F:AC  ACL MTU: 1021:8  SCO MTU: 64:1
        DOWN
        RX bytes:3414 acl:0 sco:0 events:385 errors:0
        TX bytes:66402 acl:0 sco:0 commands:385 errors:0

root@OpenWRT:~# hciconfig  hci0 up
root@OpenWRT:~# hciconfig
hci0:   Type: Primary  Bus: UART
        BD Address: 43:45:C0:00:1F:AC  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:4108 acl:0 sco:0 events:422 errors:0
        TX bytes:66841 acl:0 sco:0 commands:422 errors:0

root@OpenWRT:~# hciconfig  device
hci0:   Type: Primary  Bus: UART
        BD Address: 43:45:C0:00:1F:AC  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:4108 acl:0 sco:0 events:422 errors:0
        TX bytes:66841 acl:0 sco:0 commands:422 errors:0
```
use command ` hciconfig  hci0 up` to up BT

ophub, 2022-04-11:
ok