ophub/amlogic-s9xxx-openwrt #240: s905x mxq pro plus 无法进入系统安装

lppsuixn, 2022-04-08:
我有一台 mxq pro plus的盒子，s905x 的cpu，2g+16g配置。刷了s905x的包无法进入安装界面，直接到盒子系统。
但是刷emuelec系统替换dtb后，以及 https://pulpstone.pw/openwrt-18-06-2/ 这里的s905x openwrt都可以正常安装。
所有想了解下有debug的方法吗？该如何定位问题，谢谢~

ophub, 2022-04-08:
MXQ Pro是s905，不是s905x，这个盒子不支持写入emmc，写入必砖

lppsuixn, 2022-04-08:
这个机器有很多版本。我的是这个
https://www.geekbuying.com/item/MXQ-PRO--Android-7-1-1-Amlogic-S905X-4K-TV-BOX-2GB-16GB-385604-385604.html

ophub, 2022-04-08:
如果确定是s905x，设备刷过emuelec系统的话必须先刷会原安卓TV系统，才可以从USB启动。

lppsuixn, 2022-04-08:
emuelec 和 pulpstone openwrt 都是在u盘启动的。emmc内电视系统没有动过。
我也测试了F大的n1的系统，然后改成p212的dtb，也是启动不了。
有其他需要改的吗？比如u-boot之类的需要调吗？
谢谢。

ophub, 2022-04-08:
5.4内核的固件不需要u-boot，只有5.10以后的才需要。所以只要/boot/uEnv.txt里的dtb的文件名称正确即可。你可以测试下其他dtb，你以前在emuelec 和 pulpstone openwrt里使用的是哪个dtb，看下他的名称。
https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/build-armbian/common-files/patches/amlogic-dtb
在这里找下可以使用的，更改/boot/uEnv.txt里的dtb的名称测试。

lppsuixn, 2022-04-08:
好的，我再测试下。感谢

lppsuixn, 2022-04-08:
[image]
目前看到emuelec 和 pulpstone 里命名一致，都是gxl_p212_xxx.dtb 大小基本一致
奇怪的是 pulpstone里没有指定dtb地址
[image]

目录截图
[image]

ophub, 2022-04-08:
uEnv.ini用文本打开

closed 2022-04-10 by ophub