ophub/amlogic-s9xxx-openwrt #241: add support for official openwrt 21.02.2 stable release please!

Kopi1984, 2022-04-08:
I tried to compile OpenWRT 21.02.2 statble release, but it seems complicated to do, could you please help make this simple or just compile and push it to release of your repository?

Thank you much!
Kopi.

ophub, 2022-04-08:
You don't need to enter a word, just click to start in the Actions

closed 2022-04-08 by ophub

Kopi1984, 2022-04-10:
Thank you for your immediately support, but that's not what I asked for, I look for the official stable release of openwrt (not snapshot version). 
3 days before, I downloaded the OpenWRT rootfs file from this link: https://downloads.openwrt.org/releases/21.02.2/targets/armvirt/64/openwrt-21.02.2-armvirt-64-default-rootfs.tar.gz.
Then I tried to compile the openwrt locally with clone of your repository and the rootfs that downloaded above. The result of this compilation was the pure official openwrt stable release version, unfortunately this firmware had no amlogic service (I love this service).
Is there any method to compile pure official stable release of openwrt 21.02.2 with amlogic service for these boxes?
I hope you got what I want to describe, sorry for my bad English.

Thank you very much.
Kopi.

ophub, 2022-04-10:
https://github.com/ophub/amlogic-s9xxx-openwrt#use-github-action-to-packaging-instructions

Kopi1984, 2022-04-13:
> https://github.com/ophub/amlogic-s9xxx-openwrt#use-github-action-to-packaging-instructions
> 
> `openwrt_path`: Set the file path of openwrt-armvirt-64-default-rootfs.tar.gz , you can use a relative path such as `openwrt/bin/targets/*/*/*rootfs.tar.gz` or the network file download address. E.g `https://github.com/*/releases/*/*rootfs.tar.gz`
> 
> https://github.com/ophub/luci-app-amlogic#manual-install
> 
> If the OpenWrt you are using does not have this plugin, you can also install it manually. Use `SSH` to log in to any directory of OpenWrt system, Or in the `System menu → TTYD terminal`, Run the onekey install command to automatically download and install this plugin. `curl -fsSL git.io/luci-app-amlogic | bash`

I follow the one key install but could not access amlogic service, everytime I click on amlogic service in system menu, it showed up these information:
/usr/lib/lua/luci/dispatcher.lua:1347: module 'luci.cbi' not found:
	no field package.preload['luci.cbi']
	no file './luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi/init.lua'
	no file '/usr/lib/lua/luci/cbi.lua'
	no file '/usr/lib/lua/luci/cbi/init.lua'
	no file './luci/cbi.so'
	no file '/usr/lib/lua/luci/cbi.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:1347: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:1000: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

I just dont understand what is that, could you please give me a little help about this issue?

Thank you.
kopi.