ophub/amlogic-s9xxx-openwrt #242: 能提供对S905X3芯片的Tox1的支持吗？

lt16caep, 2022-04-11:
利用U盘启动TOX1后，发现有线网卡无法驱动，换用Ugoos X3，x96-air等S905X3盒子的DTB后均无法驱动有线网卡。
后来从TOX1的刷机包中提取出DTS文件后，并以Ugoos X3，x96-air等盒子的DTS文件（利用dtc从DTB转换到了DTS格式）作为参考，但发现TOX1原生安卓的DTS格式与Openwrt的差异很大。本人并非IT背景，因此无从下手。
虽然尝试对ethernet@ff3f0000部分进行了修改，并转换为dtb文件后，发现该dtb文件无法找到网卡。以下是我提取的TOX1原生安卓dts文件，以及转换后的DTB文件。能请你帮忙修改出能够支持TOX1网卡的DTB文件吗？谢谢
[sm1-tox1-4g.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/8464197/sm1-tox1-4g.zip)

ophub, 2022-04-12:
安卓的不是主线内核，不能使用
默认的s905x3固件不修改，启动后有网络没。

lt16caep, 2022-04-12:
若采用刷机包中提供的其他S905X3的DTB文件，openwrt会发现网卡，但似乎因为参数不对无法正常使用。启动时的log信息如下：
[   25.694711] meson8b-dwmac ff3f0000.ethernet eth0: validation of rmii with support 0000000,00000000,00006280 and advertisement 0000000,00000000,00006280 failed: -22
[   25.708016] meson8b-dwmac ff3f0000.ethernet eth0: no phy at addr -1
[   25.713920] meson8b-dwmac ff3f0000.ethernet eth0: stmmac_open: Cannot attach to PHY (error: -19)
[   25.723100] br-lan: port 1(eth0) entered blocking state
[   25.727804] br-lan: port 1(eth0) entered disabled state
[   25.733284] device eth0 entered promiscuous mode

使用刷机包中的其他S905X3固件（包括meson-sm1-sei610.dtb及meson-sm1-sei610-ethfix.dtb），同样能引导openwrt（但网卡不工作），反馈的滚动信息与此类似。

进入命令行后输入ethtool -s eth0 autoneg off可以使有线网卡工作在百兆状态下。此时，能够通过浏览器进入Openwrt的网页管理页面。但不知道是否是因为命令关闭了网卡的自适应，导致通过网页端调整任何跟网络有关的设置后，都会导致有线网络掉线，需要重新插拔网线才能恢复网络连通的状态。此外，虽然此时盒子能够正常工作，但似乎极不稳定，一段时间（时间长短不固定）后会出现有线网络掉线，且重新插拔网线无法恢复的现象。 也可以换用其他dtb文件，如meson-sm1-x96-air-gbit.dtb等，情况与此类似。

lt16caep, 2022-04-12:
压缩包中的dtb文件，是以meson-sm1-x96-air-gbit.dtb为基础修改的。我看不太懂代码，不清楚具体需要修改哪些位置。只是凭借朴素的感觉，参照安卓的dts修改了ethernet@ff3f0000中的部分reg、regname、phandle等字段的参数。
能帮忙修改一个支持tox1的dtb文件吗？谢谢

lt16caep, 2022-04-12:
设备是千兆网卡rtl8211f，这个设备老毛子用的比较多。
我测试一下5.4的固件后来反馈

closed 2022-04-13 by ophub

lt16caep, 2022-04-27:
采用openwrt_s905x3_k5.15.34_2022.04.18.2353.img及meson-sm1-hk1box-vontar-x3.dtb启动后，在/etc/rc.local中添加如下命令 sleep 40 && ethtool -s eth0 autoneg on，之后再按照网络环境修改/etc/config/network的LAN IP。
目前已正常使用4天多。