ophub/amlogic-s9xxx-openwrt #243: (s905x3 tx3 4+64) vfd依然不显示时间，只显示boot

13528943945, 2022-04-12:
fork了最新的仓库，用openwrt-21.02的yml脚本云编译。刷机后，终端运行openwrt-led，选16，提示设备打开失败。日志如下：

[  113.097735] OpenVFD: Version: V1.4.1
[  113.097743] OpenVFD: vfd_gpio_clk:		Empty.
[  113.097746] OpenVFD: vfd_gpio_dat:		Empty.
[  113.097748] OpenVFD: vfd_gpio_stb:		Empty.
[  113.097753] OpenVFD: vfd_gpio0:		#0 = 0x00; #1 = 0x00; #2 = 0xFF; 
[  113.097756] OpenVFD: vfd_gpio1:		#0 = 0x00; #1 = 0x00; #2 = 0xFF; 
[  113.097760] OpenVFD: vfd_gpio2:		#0 = 0x00; #1 = 0x00; #2 = 0xFF; 
[  113.097764] OpenVFD: vfd_gpio3:		#0 = 0x00; #1 = 0x00; #2 = 0xFF; 
[  113.097766] OpenVFD: vfd_gpio_protocol:	#0 = 0x00; #1 = 0x00; 
[  113.097768] OpenVFD: vfd_chars:		Empty.
[  113.097770] OpenVFD: vfd_dot_bits:		Empty.
[  113.097772] OpenVFD: vfd_display_type:	Empty.
[  113.097779] OpenVFD: Detected gpio chips:	aobus-banks, periphs-banks.
[  113.097781] OpenVFD: Failed to verify VFD configuration file, attempt using device tree as fallback.
[  113.104460] OpenVFD: openvfd_gpio_clk pin entry not found
[  113.104462] OpenVFD: openvfd_gpio_dat pin entry not found
[  113.104464] OpenVFD: openvfd_gpio_stb pin entry not found
[  113.104467] OpenVFD: openvfd_gpio0 pin entry not found
[  113.104469] OpenVFD: openvfd_gpio1 pin entry not found
[  113.104471] OpenVFD: openvfd_gpio2 pin entry not found
[  113.104472] OpenVFD: openvfd_gpio3 pin entry not found
[  113.104475] OpenVFD: can't find openvfd_chars list, falling back to defaults.
[  113.104477] OpenVFD: chars_prop = 0000000000000000
[  113.111736] OpenVFD: can't find openvfd_dot_bits list, falling back to defaults.
[  113.111738] OpenVFD: dot_bits_prop = 0000000000000000
[  113.118820] OpenVFD: display.type = 0, display.controller = 0, pdata->dev->dtb_active.display.flags = 0x00
[  113.118823] OpenVFD: can't request gpio of gpio_clk
[  113.118843] openvfd: probe of openvfd failed with error -22

ophub, 2022-04-12:
cat /boot/uEnv.txt

13528943945, 2022-04-12:
千兆的，就是这个dtb。越来越感觉到，是我自定义.config编译出来的rootfs底包缺少某些组件。驱动这个vfd需要哪些必须的组件，可以明确一下吗？感谢！

ophub, 2022-04-12:
只要dtb里面有即可，你先下载我的试一下再说

13528943945, 2022-04-12:
你的固件确实可以让vfd显示时间。那么，我们之间唯一的区别就在于使用的.config不同了。

ophub, 2022-04-12:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/bc2f2f430b66c70ff0483047cfd55f7dc849c12d
两个条件：
1. 有/sbin/kmod及相关的几个软链接
2. dtb里面有openvfd

13528943945, 2022-04-12:
[image]

13528943945, 2022-04-12:
安装上述几个工具包之后，vfd终于可以正常显示时间了。多谢大神不厌其烦的指点！

ophub, 2022-04-13:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/make#L537-L543

你安装的这几个包实际上都在打包的时候删除了，都软连接到了kmod上了。

13528943945, 2022-04-13:
意味着一个kmod搞定，那几个可以不装?

ophub, 2022-04-13:
对，装了也被删除了，你看看上面发你的代码

13528943945, 2022-04-13:
好像小时部分不对劲，现在是9:04，vfd显示01 04

ophub, 2022-04-13:
登录系统菜单，同步下盒子的时间

13528943945, 2022-04-13:
> 登录系统菜单，同步下盒子的时间

本地盒子时间正常，vfd不正常

ophub, 2022-04-13:
系统菜单--系统---基本设置--同步浏览器时间---保存&应用
openwrt-led 0
openwrt-led 16

13528943945, 2022-04-13:
还是不对，依然01 13而现在是09 13

ophub, 2022-04-13:
错了，是17

13528943945, 2022-04-13:
17更加不对，调用了x96的

13528943945, 2022-04-13:
[image]

13528943945, 2022-04-13:
[image]

ophub, 2022-04-13:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/3a58f4798fbd800687ba70eeb84f218c209af05d

刚才发现，上传调整排序的时候只改了armbian的，忘了给OpenWrt调整了，你看看你本地的代码，根据实际提示使用

13528943945, 2022-04-13:
[image]

ophub, 2022-04-13:
[image]

13528943945, 2022-04-13:
[image]

ophub, 2022-04-13:
https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/build-armbian/common-files/rootfs/usr/share/openvfd

你同步最新的代码，conf里的tx3文件上周有修改

13528943945, 2022-04-13:
昨天刚刚fork的

ophub, 2022-04-13:
重启试试

13528943945, 2022-04-13:
不行，相差8小时

ophub, 2022-04-13:
时区不对，系统里调整

13528943945, 2022-04-13:
[image]

ophub, 2022-04-13:
`cd /sbin && [ -x kmod ] && { kmod_list="depmod insmod lsmod modinfo modprobe rmmod"; for ki in ${kmod_list}; do   rm -f ${ki};  ln -sf kmod ${ki}; done } && echo "ok"`

ophub, 2022-04-13:
ping baidu.com
看看网络通不

13528943945, 2022-04-13:
复制整句在终端执行后，显示ok。然后openwrt-led 0，再openwrt-led 16。还是相差八小时。网络是通的

ophub, 2022-04-13:
ping或者nslookup截图，我看看网络是不是通了，时区是否生效了

13528943945, 2022-04-13:
[](url
[image]
)

13528943945, 2022-04-13:
[image]

ophub, 2022-04-13:
`cat /boot/uEnv.txt`

13528943945, 2022-04-13:
meson-sm1-tx3-qz.dtb

13528943945, 2022-04-13:
[image]

13528943945, 2022-04-13:
[image]

ophub, 2022-04-13:
https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-docs/led_screen_display_control.md

咱们两个的显示不知道为啥不一样，根据这个说明里的参数你调整下顺序测试。
我在f大的群里看见有人使用tx3，你问下谁的正常，要他的tx3.conf测试下。

13528943945, 2022-04-13:
好的，我试试，有结果再反馈。谢谢！

closed 2022-04-13 by ophub

ophub, 2022-04-14:
[zoneinfo.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/8487082/zoneinfo.zip)
`opkg install *.ipk`
The problem is found. The package zoneinfo is missing. After installation, synchronize the time in the browser, and it can be displayed correctly.

13528943945, 2022-04-14:
是的，安装完两个ipk之后，重启，vfd时间正常了。多谢大佬！