ophub/amlogic-s9xxx-openwrt #248: 大神，我把打包脚本弄到本地ubuntu出现以下错误，有影响吗？

281677160, 2022-04-25:
[image]

ophub, 2022-04-25:
这太影响了，打包完了完全不能用。
你的本地文件不安整。踩高跷

281677160, 2022-04-25:
呃，好久没搞了，以前你没改版的时候就是这样没问题的，最近你的脚本好像改的挺多的，哈哈

281677160, 2022-04-25:
没改版的时候本地打包没问题

ophub, 2022-04-25:
很多文件在另一个armbian库里，要从那个仓库下载

281677160, 2022-04-25:
就是把这两个，覆盖到这个仓库的同名文件
[image]
吗

281677160, 2022-04-25:
感谢大佬，我把你https://github.com/ophub/amlogic-s9xxx-armbian这个仓库的build-armbian里面的所有文件丢一个文件夹，然后把[compile-kernel]文件下载下来也丢进去，最后把https://github.com/ophub/amlogic-s9xxx-openwrt这个仓库的compile-kernel文件覆盖过去，打包好像没问题了，就是不知道对
[image]
不对

ophub, 2022-04-25:
你图上的下载说明可以正常从github下载文件了。所以不用复制了，打包脚本会自动处理好。
没报错就可以用了

closed 2022-04-25 by ophub