ophub/amlogic-s9xxx-openwrt #251: error 403

elsangithub, 2022-05-07:
Run ncipollo/release-action@v1
Error: Error 403: Resource not accessible by integration

ophub, 2022-05-07:
https://github.com/ncipollo/release-action

It can be accessed normally, you can try again

closed 2022-05-07 by ophub

elsangithub, 2022-05-08:
still the same

Run ncipollo/release-action@v1
Error: Error 403: Resource not accessible by integration

ophub, 2022-05-08:
https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/2286380871

ophub, 2022-05-08:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#2-set-the-privacy-variable-github_token

elsangithub, 2022-05-08:
I've done it several times in the same way but it's the same.. 1 month ago everything was very smooth.  but now the file can't be release.  I also tried it with a different account but it's the same.