ophub/amlogic-s9xxx-openwrt #252: Error: Error 403: Resource not accessible by integration

nikoharo, 2022-05-09:
我尝试编译很多次了，好像是没赋予fork写入权限？

elsangithub, 2022-05-09:
I also found the same thing

ophub, 2022-05-09:
以前没发现有权限问题，你知道怎么赋权？

ophub, 2022-05-09:
I read the instructions, token is not required, this parameter has been removed, I tested it and it can be used. You sync the source code and try again

nikoharo, 2022-05-09:
> 以前没发现有权限问题，你知道怎么赋权？ https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/README.cn.md#2-%E8%AE%BE%E7%BD%AE
%E9%9A%90%E7%A7%81%E5%8F%98%E9%87%8F-github_token
> 
> 你给自己的仓库有赋予token权限了没

[image]
[image]

后面参考了这篇文章修改参数也还是不行

ophub, 2022-05-09:
同步刚才的修改，我删除token了，测试可用

nikoharo, 2022-05-09:
> 同步刚才的修改，我删除了token了，测试可用 https://github.com/ophub/amlogic-s9xxx-openwrt/actions/runs/2292952862

ok,我试试

closed 2022-05-09 by ophub

kousyougi, 2022-05-10:
刚发现最近新fork的repositories预设没有写入Release的权限。很久以前fork预设开放写入权限。
到下面设定一下就不会跳出错误了。

[image]

ophub, 2022-05-10:
https://github.com/alex-page/github-project-automation-plus/issues/39
我在这个帖子里找到了解决方案，说要创建的密钥命名是：GH_TOKEN，我测试了有效。
这个问题从昨天发现，我测试了一晚上，各种Actions权限修改也没解决，就在你发帖前我刚刚测试成功。你试试我的这种做法。
[image]

elsangithub, 2022-05-10:
OK.. i try

elsangithub, 2022-05-10:
success, thank you everyone