ophub/amlogic-s9xxx-openwrt #258: Unable to build the latest kernel files

dinhchinh82, 2022-05-23:
I've just forked the latest kernel and built via Github action but got the error 
[image]

@ophub could you give me the suggestion to fix?

Thank you.

ophub, 2022-05-23:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#2-set-the-privacy-variable-github_token

2. Set the privacy variable GitHub_TOKEN
3. Fork repository and set GH_TOKEN

github.com has modified the default `fork` settings, you need to refer to the instructions and assign these permissions to `Actions` before you can use them

dinhchinh82, 2022-05-23:
> github.com has modified the default `fork` settings, you need to refer to the instructions and assign these permissions to `Actions` before you can use them

Yes. I just found this reason. Thank you for your help.

closed 2022-05-23 by dinhchinh82