ophub/amlogic-s9xxx-openwrt #260: Nexbox A1 Wireless Support

kodokskoprol, 2022-05-29:
I have device branded as A95X, it has the same form factor and specification as Nexbox A1
How to enable wireless function?

the wireless module is Fn-Link 8223A-SR C07001 1727

**https://www.gearbest.com/tv-box-mini-pc/pp_424843.html**
[image]
[image]

ophub, 2022-05-29:
I consulted flippy, this is the `qca9377` chip, it is recommended to try the `ath10k driver.` I newly added this driver today, you download today's firmware and try

kodokskoprol, 2022-05-29:
tried all s912 lastest firmware, wireless still not showing

[image]

ophub, 2022-05-30:
[meson-gxm-a95x.dtb.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/8795710/meson-gxm-a95x.dtb.zip)

Download and decompress the meson-gxm-a95x.dtb file and upload it to the /boot/dtb/amlogic directory. Modify the name of the dtb file in `/boot/uEnv.txt` to `meson-gxm-a95x.dtb`, restart after the modification, try whether the wireless can be used, and report the result.

kodokskoprol, 2022-05-30:
lastest firmware & a95x dtb unfortunately still not working
[image]

ophub, 2022-05-30:
[image]

[image]

[image]

GPIOX_17 corresponds to a pin on the cpu and needs to be measured with a multimeter, or you can find the official design schematic of this motherboard

[meson-gxm-a95x.dts.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/8796574/meson-gxm-a95x.dts.zip)

This is the source file of the dtb added for your device.

```
&uart_A {
		status = "okay";
		pinctrl-0 = <&uart_a_pins>, <&uart_a_cts_rts_pins>;
		pinctrl-names = "default";
		uart-has-rtscts;

		bluetooth {
				compatible = "qcom,qca9377-bt";
				shutdown-gpios = <&gpio GPIOX_17 GPIO_ACTIVE_HIGH>;
				max-speed = <2000000>;
				clocks = <&wifi32k>;
				clock-names = "lpo";
		};
};
```

The value of `GPIOX_17` needs to determine what your pin is.

ophub, 2022-05-30:
Is the wired network available? Is the network port of this device Gigabit or 100M?

kodokskoprol, 2022-05-30:
the wired ethernet is gigabit and working

kodokskoprol, 2022-05-30:
i have multitester, but i didn't know how to measure the GPIO

ophub, 2022-06-01:
Need to remove the cpu, it is more troublesome, stop measuring, use the wired network

kodokskoprol, 2022-06-01:
Flippy kernel does not have `u-boot.ext`.

Ophub kernel using `u-boot.ext`. file that work with this kernel is
`u-boot-p201.bin`
`u-boot-p212.bin`
`u-boot-s905.bin`
i have to manually rename those file into `u-boot.ext` , because `u-boot.ext` that came with firmware didn't work

but both kernel showing same result in the terminal
[image]
[image]

ophub, 2022-06-01:
`u-boot-p212.bin`
This can be used normally under the 5.10 and 5.15 kernels? Then make sure it is this.

ophub, 2022-06-01:
https://github.com/unifreq/linux-5.10.y/blob/efde4bfba2724ae4c2e2afdbb53813f45f4c8829/arch/arm64/boot/dts/amlogic/meson-gxl-s905d-sml5442tw.dts#L74

I found one in the dtb directory that uses the same chip as your qca9377, and its pin is also GPIOX_17. This value is also the setting in the file I sent you before.

Where can I find the correct value for this.

ophub, 2022-06-01:
[meson-gxm-a95x-a.dtb.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/8812129/meson-gxm-a95x-a.dtb.zip)
[meson-gxm-a95x-b.dtb.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/8812130/meson-gxm-a95x-b.dtb.zip)

Can you try these 2 dtbs to use wireless again?
Upload to the `/boot/dtb/amlogic` directory
Then `vi /boot/uEnv.txt` modify the dtb name inside

kodokskoprol, 2022-06-02:
tested the dtb unfortunately not working

danboid, 2022-06-02:
Those dtbs don't boot at all on my X96 Air Q1000 and my X96 Max+ 64 GB.

danboid, 2022-06-02:
The latest Manjaro linux-aml kernel package is at 5.13. Wifi works with 5.13 but it has HDMI issues. I would recommend any X96 users using Manjaro use my 5.12 kernel packages instead because they have working HDMI

https://github.com/danboid/meson-sm1-sei610-qca9377-bt/releases/tag/kernel

danboid, 2022-06-02:
@ophub 

I'm sorry, I gave you the old manjaro ARM amlogic kernel repo address earlier. The current one is:

https://gitlab.manjaro.org/manjaro-arm/packages/core/linux-odroid

I have tested Manjaros 5.17 odroid kernel on my X96 Max+ 64 GB and my X96 Air with their meson-g12a-x96-max.dtb and the both onboard wifi and ethernet works.

ophub, 2022-06-03:
@danboid 
What is the name of your device? soc? NIC chip?

ophub, 2022-06-03:
In the address you sent, no modification was found for meson-g12a-x96-max.dtb or related include files. Do you know which patch he used?

danboid, 2022-06-03:
s905x3.

What about the kernel network config options? Are they identical? I'll try and have a look later but I'm busy most of the day today.

danboid, 2022-06-03:
@ophub 

If you send me your address I'll send you my spare X96 Air Q1000. I'd like to see s9xxx Armbian have working wifi and bluetooth for these boxes.

ophub, 2022-06-03:
This is the default setting for the mainline kernel：

https://github.com/unifreq/linux-5.15.y/blob/main/arch/arm64/boot/dts/amlogic/meson-g12a-x96-max.dts

Know which value to change, create a new dtb like this. Just modify those few lines：

https://github.com/unifreq/linux-5.15.y/blob/main/arch/arm64/boot/dts/amlogic/meson-g12a-x96-max-rmii.dts

I can't find it in the URL you posted today. You can look for it. Or decompile the dtb you can use, and compare it with the mainline to see what has been adjusted.

closed 2022-06-04 by ophub