ophub/amlogic-s9xxx-openwrt #262: X96max+ with IP1001M gigbitye ethernet not working

manalwong, 2022-05-31:
Hi Ophub, the lastest release seems to be included the driver for IP1001M right?

However I just test with my box, the device can boot up but no eth0 can not came up.

I had changed the uEnv file to dtb 'meson-sm1-x96-max-plus-ip1001m.dtb' with no lucky, please advise.

the error message seems to said: 'ff3f0000.ethernet xxx cannot attach to phy (error: -19)'

ophub, 2022-06-01:
It was reported that there were contrary test results

manalwong, 2022-06-01:
So it can be supported or not?

ophub, 2022-06-01:
[meson-sm1-x96-max-plus-ip1001m.dtb.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/8814014/meson-sm1-x96-max-plus-ip1001m.dtb.zip)

Download this file and unzip it, then overwrite /boot/dtb/amlogic/meson-sm1-x96-max-plus-ip1001m.dtb with the same name file, restart and try

manalwong, 2022-06-01:
Wow! Big ophub, you are rock! It works!!!

From iperf test:
Upload: about 860 Mbps
Download: about 826 Mbps

[Test result](url
[x96max+ iperf test result.txt](https://github.com/ophub/amlogic-s9xxx-openwrt/files/8814739/x96max%2B.iperf.test.result.txt)
)

manalwong, 2022-06-01:
Anything else informations that you want from my box?

ophub, 2022-06-01:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/50dead3c58edbe97022f4fcac8271a32035ffd47/amlogic-s9xxx/common-files/rootfs/etc/model_database.txt#L41

The dtb file was replaced in the usb. If it can be driven normally after restarting, is it convenient to write emmc to test it?

If you need to keep the Android system in emmc, you can run the openwrt system in the usb first and use openwrt-ddbr for backup. You can use this command to restore Android later.

closed 2022-06-04 by ophub

Zas09, 2022-11-17:
> Wow! Big ophub, you are rock! It works!!!
> 
> From iperf test: Upload: about 860 Mbps Download: about 826 Mbps
> 
> [Test result] [x96max+ iperf test result.txt](https://github.com/ophub/amlogic-s9xxx-openwrt/files/8814739/x96max%2B.iperf.test.result.txt)

Hello, what is kernel did u flashed ur box? I have same box x96max plus ai2 , ip1001m ...

ophub, 2022-11-17:
https://github.com/ophub/amlogic-s9xxx-armbian/blob/d3fcaac61334983605d2d4a2475354309cbda626/build-armbian/common-files/rootfs/etc/amlogic_model_database.conf#L94

Zas09, 2022-12-05:
> meson-sm1-x96-max-plus-ip1001m.dtb
> 
> https://github.com/ophub/amlogic-s9xxx-armbian/blob/d3fcaac61334983605d2d4a2475354309cbda626/build-armbian/common-files/rootfs/etc/amlogic_model_database.conf#L94

Thanks