ophub/amlogic-s9xxx-openwrt #265: [Enchantment] build selector

rapdodge, 2022-06-13:
Dear ophub, can you make build selector when we want to build openwrt?

I mean, we can choose SNAPSHOT or latest stable by openwrt. Because when i want to installed another packages after running it, it says kernel is not compatible

[image]

Thanks

ophub, 2022-06-13:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config

Different branches of different source codes have different configuration files. Here are the configuration files of the two official versions. Other branches can be created and compiled in this way.

rapdodge, 2022-06-13:
if i'm not forget last night i try to change to openwrt-21.02
and its still SNAPSHOT

and i destroyed the repo and reforking again...

ophub, 2022-06-13:
https://github.com/0118Add/amlogic-s9xxx-openwrt/tree/main/router-config

I checked the fork list, there is also the previous configuration of openwrt-21.02 in this repository, you can make a copy. When compiling natively every time a fresh compile. Or use github's actions to compile and avoid caching.

rapdodge, 2022-06-13:
Will take a look into it

rapdodge, 2022-06-13:
Update... Its still SNAPSHOT, and, other packages that i want to install keep says not compatible...
So, i think i should rebuilding it every i want to get some packages 😅

[image]

Thanks for your hardwork!

closed 2022-06-13 by rapdodge

rapdodge, 2022-06-13:
Btw @ophub, can you tell me how to generate router-config/.config file?

Is there any tools or something like that to generate that file?

ophub, 2022-06-13:
https://github.com/openwrt/openwrt#quickstart

Run `make menuconfig` to select your preferred configuration for the toolchain, target system & firmware packages. `save` and generate `.config`