ophub/amlogic-s9xxx-openwrt #268: [Request] Adding workflows rebuild openwrt like armbian

rapdodge, 2022-06-17:
Dear ophub,

Can you make an workflows for github actions to rebuild openwrt using vanilla openwrt that downloaded like from https://downloads.openwrt.org/releases/21.02.3/targets/armvirt/64/ <- here

I'm requesting this, because i want to get stable openwrt (not snapshot) to make installing package that require kernel should be same is easier...

Ref: https://github.com/ophub/amlogic-s9xxx-openwrt/issues/241

Much thanks

ophub, 2022-06-17:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#1011-required-options-for-openwrt

Since the official precompiled firmware lacks necessary dependencies, there are some problems in installation and use, and it cannot be used. You need to add these dependencies yourself to compile the appropriate crab

rapdodge, 2022-06-17:
Noted, there is an imagebuilder official from openwrt to making rootfs

https://downloads.openwrt.org/releases/21.02.3/targets/armvirt/64/openwrt-imagebuilder-21.02.3-armvirt-64.Linux-x86_64.tar.xz

Can you please translating it to can be use github actions?

ophub, 2022-06-17:
Implementing your needs is simple.
You send the official configuration, I will see if these dependencies are met, otherwise the firmware produced cannot be used normally.
Not everyone will read the documentation, and making firmware that doesn't work can cause a lot of problems.

rapdodge, 2022-06-17:
Trying to get really vanilla openwrt

`wget https://downloads.openwrt.org/releases/21.02.3/targets/armvirt/64/openwrt-imagebuilder-21.02.3-armvirt-64.Linux-x86_64.tar.xz`

`tar -J -x -f openwrt-imagebuilder-21.02.3-armvirt-64.Linux-x86_64.tar.xz`

`cd openwrt-imagebuilder-21.02.3-armvirt-64.Linux-x86_64`

`make image PROFILE="Default" PACKAGES="\
perl-http-date perlbase-getopt perlbase-time perlbase-unicode perlbase-utf8 \
blkid fdisk lsblk parted attr btrfs-progs chattr dosfstools e2fsprogs f2fs-tools \
f2fsck lsattr mkf2fs xfs-fsck xfs-mkfs bsdtar pigz bash gawk getopt losetup pv tar \
uuidgen coremark coreutils coreutils-base64 coreutils-nohup kmod-brcmfmac kmod-brcmutil \
kmod-cfg80211 kmod-mac80211 hostapd-common wpa-cli wpad-basic iw" FILES="files"`

^[Ref](https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#1011-required-options-for-openwrt)

[image]

Rootfs file https://transfer.sh/MOqJcr/openwrt-21.02.3-armvirt-64-default-rootfs.tar.gz

Github link: https://github.com/rapdodge/amlogic-s9xxx-openwrt/releases/tag/openwrt_s9xxx_official_06.17.2224

Above is just an barebone openwrt from [Ref](https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#1011-required-options-for-openwrt), but its nice to having other packages like luci, wireguard, etc

Will testing it right now...

rapdodge, 2022-06-17:
Its on, and its working, STB can detect NIC plugged and unplug, but i cant setting for the internet, because i'm not familiar with the CLI...

[image]

[image]

[image]

ophub, 2022-06-17:
How to set up the network: ssh, `vi /etc/config/network`

rapdodge, 2022-06-17:
> This is the first time I see this operation. Can I directly add dependency packages and plug-ins to the firmware? How long does it take you to append these dependencies?
> 
> Do you want one like this? https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/.github/workflows/rebuild-armbian.yml
> 
> Download and compile directly from the given address?
> 
> [image]

its not long, it just downloading it and configuring it...

Some screenshot
[image]

[image]

[image]

rapdodge, 2022-06-17:
let me rebuilding it with luci... will keep you update

ophub, 2022-06-17:
I asked you the question above, is the function you want the same as the one I posted?

rapdodge, 2022-06-17:
> I asked you the question above, is the function you want the same as the one I posted?

yes, i'd like to have workflow like rebuild armbian :D

ophub, 2022-06-17:
ok, very simple

rapdodge, 2022-06-17:
Btw, if just rootfs without depedency that should be included on [ref](https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config#1011-required-options-for-openwrt) if im not forgot, it cant be detect the NIC is plug or unplug...

I already doing this
https://github.com/rapdodge/amlogic-s9xxx-openwrt/blob/fd6781416d848c510da91fe809925227880d9dc6/.github/workflows/use-releases-file-to-packaging.yml#L112

rapdodge, 2022-06-17:
let me try again... direct using openwrt default rootfs... but, before that, let me check using latest imagebuild with luci

rapdodge, 2022-06-17:
Ok @ophub, i tested it with just openwrt base rootfs without customization, and it booting, but, NIC didnt respond...

ophub, 2022-06-17:
This is the first time I have seen your compilation method. I used to build the tar.gz file completely from scratch. I have to learn to use this method, it seems very convenient, at least 3 hours of compilation time is omitted. If it's reliable, it's a great way.

rapdodge, 2022-06-17:
Some other reference
https://openwrt.org/docs/guide-user/additional-software/imagebuilder

Looking forward into it :D

zakeru46, 2022-06-17:
do you have amlogic service included? tried this imagebuilder method it seems need to be compiled first. I thought ophub's repack script has taken care of it.

ophub, 2022-06-17:
The process of installing dependencies, does it support downloading third-party ipk installation?

rapdodge, 2022-06-18:
Actually, I don't know...

ophub, 2022-06-18:
yes

rapdodge, 2022-06-18:
Ok, let me test it with my build openwrt barebone with luci, will update you asap

rapdodge, 2022-06-18:
It's ok, no problem when installing it

[image]

[image]

ophub, 2022-06-18:
Can a third-party ipk be installed during imagebuild?

rapdodge, 2022-06-18:
> Can a third-party ipk be installed during imagebuild?

I dont know, to be honest...

ophub, 2022-06-18:
> And i think it more nice when the workflows can be added multi options what we want to build like just 21.02.3, 21.02.2, 21.02.1, etc...

> Because the link its likely same
https://downloads.openwrt.org/releases/ **version**/targets/armvirt/64/openwrt-imagebuilder...tar.xz

Send a few addresses like yours, let me see the rules, how to add a suitable drop-down list

rapdodge, 2022-06-18:
Dear @ophub,

I think its 3rdparty ipk can be installed during imagebuild

Screenshot:

[image]

You should place custom ipk on packages folder inside the imagebuilder directory and adding it on make

[image]

`make image PROFILE="Default" PACKAGES="\
perl-http-date perlbase-getopt perlbase-time perlbase-unicode perlbase-utf8 \
blkid fdisk lsblk parted attr btrfs-progs chattr dosfstools e2fsprogs f2fs-tools \
f2fsck lsattr mkf2fs xfs-fsck xfs-mkfs bsdtar pigz bash gawk getopt losetup pv tar \
uuidgen coremark coreutils coreutils-base64 coreutils-nohup kmod-brcmfmac kmod-brcmutil \
kmod-cfg80211 kmod-mac80211 hostapd-common wpa-cli wpad-basic iw \
liblucihttp liblucihttp-lua luci luci-app-firewall luci-app-opkg luci-base luci-lib-base \
luci-lib-ip luci-lib-jsonc luci-lib-nixio luci-mod-admin-full luci-mod-network \
luci-mod-status luci-mod-system luci-proto-ipv6 luci-proto-ppp \
luci-theme-bootstrap luci-compat luci-app-amlogic" FILES="files"`

rapdodge, 2022-06-18:
> Send a few addresses like yours, let me see the rules, how to add a suitable drop-down list

dropdown i mean is like this

[image]

And adding latest major version like:
21.02.3 -> https://downloads.openwrt.org/releases/21.02.3/targets/armvirt/64/openwrt-imagebuilder-21.02.3-armvirt-64.Linux-x86_64.tar.xz

19.07.10 -> https://downloads.openwrt.org/releases/19.07.10/targets/armvirt/64/openwrt-imagebuilder-19.07.10-armvirt-64.Linux-x86_64.tar.xz

18.06.9 -> https://downloads.openwrt.org/releases/18.06.9/targets/armvirt/64/openwrt-imagebuilder-18.06.9-armvirt-64.Linux-x86_64.tar.xz

[image]

rapdodge, 2022-06-18:
Testing using: https://github.com/rapdodge/amlogic-s9xxx-openwrt/releases/tag/openwrt_s9xxx_official_06.18.0914

It's booting: yes
It's can detect NIC: yes
It's have amlogic service: yes

ophub, 2022-06-18:
https://github.com/ophub/amlogic-s9xxx-openwrt/tree/main/router-config/openwrt-imagebuilder

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/.github/workflows/build-openwrt-with-imagebuilder.yml

You test, continue to add complete packages and luci-app, luci-theme

rapdodge, 2022-06-18:
Noted, will testing it when I'm back home

ophub, 2022-06-18:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/openwrt-imagebuilder/imagebuilder.sh

When using locally, just run `./imagebuilder.sh <branch>` directly, for example: `./imagebuilder.sh 21.02.3`

rapdodge, 2022-06-18:
Building is under 20 mins
[image]

Wonderful!

rapdodge, 2022-06-18:
Testing: https://github.com/rapdodge/amlogic-s9xxx-openwrt/releases/tag/openwrt_s9xxx_imagebuilder_21.02.3_06.18.2012

It's booting: yes
It's can detect NIC: yes
It's have amlogic service: yes

Some screenshots

[image]

[image]

[image]

This request i think can be closed

rapdodge, 2022-06-18:
Thanks for your hardwork!

closed 2022-06-18 by rapdodge