ophub/amlogic-s9xxx-openwrt #271: Can't install openwrt in EMMC

Roxy09099, 2022-07-02:
My devices: my device: X92, 3GB ram, 16gb rom, 1Gbps lan.
No emmc can be found to install the openwrt system
lan port got full 1000mbps everything is quite ok for me but i cant install it to EMMC memory whether kernel 5.4.y or 5.15.y

I need your help. if you have Telegram can you help me via chat? that would be more convenient on github. Think of this as a paid support service. i will pay for that.
My Telegram: https://t.me/kogoro6996
[image]

ophub, 2022-07-02:
`lsblk`
see what your emmc is
What is the `soc` of your box, install it according to the configuration you use to boot into the usb.
First use the `5.15 kernel firmware` test.

This is free software, there are no paid services. You can check the relevant operating instructions.

Roxy09099, 2022-07-02:
yes i used 5.15.15 firmware. my box is using soc s912

Roxy09099, 2022-07-02:
https://github.com/ophub/amlogic-s9xxx-openwrt/issues/196
I have a reference from this similar device use case. i also flashed stock android 6 firmware before using usb with openwrt installer

Roxy09099, 2022-07-02:
lsblk
[image]

Roxy09099, 2022-07-02:
> `lsblk` see what your `emmc name` is What is the `soc` of your box, install it according to the configuration you use to boot into the usb. First use the `5.15 kernel firmware` test.
> 
> This is free software, there are no paid services. You can check the relevant operating instructions.
yes i used 5.15.15 firmware. my box is using soc s912
emmc name: mmcblk2

Roxy09099, 2022-07-02:
> `lsblk` see what your `emmc name` is What is the `soc` of your box, install it according to the configuration you use to boot into the usb. First use the `5.15 kernel firmware` test.
> 
> This is free software, there are no paid services. You can check the relevant operating instructions.

yes i used 5.15.15 firmware. my box is using soc s912
Emmc name: mmcblk2

ophub, 2022-07-03:
[image]

The general emmc has the `mmcblk?boot0` partition, which is used to judge the name of the emmc in the script

https://github.com/ophub/luci-app-amlogic/blob/6113be1b39a63863531df3b478cd0049f8a00058/luci-app-amlogic/root/usr/sbin/openwrt-install-amlogic#L69

https://github.com/ophub/luci-app-amlogic/blob/6113be1b39a63863531df3b478cd0049f8a00058/luci-app-amlogic/root/usr/sbin/openwrt-update-amlogic#L45

Roxy09099, 2022-07-03:
> [image]
> 
> The general emmc has the `mmcblk?boot0` partition, which is used to judge the name of the emmc in the script
> 
> https://github.com/ophub/luci-app-amlogic/blob/6113be1b39a63863531df3b478cd0049f8a00058/luci-app-amlogic/root/usr/sbin/openwrt-install-amlogic#L69
> 
> ```
> # local: /usr/sbin/openwrt-install-amlogic
> # line 69, change into:
> install_emmc="$(lsblk -l -o NAME | grep -oE '(mmcblk[0-9])' | sed "s/boot0//g")"
> ```
> 
> https://github.com/ophub/luci-app-amlogic/blob/6113be1b39a63863531df3b478cd0049f8a00058/luci-app-amlogic/root/usr/sbin/openwrt-update-amlogic#L45
> 
> ```
> # loacal: /uar/sbin/openwrt-update-amlogic
> # line 45, change into:
> if lsblk -l -o NAME | grep "${EMMC_NAME}" >/dev/null; then
> ```
> 
> If you know the rescue plan of the device and have the corresponding Android TV system firmware, for example, if you try to write emmc, you can delete `?boot0` and test it

I don't know how to compile it myself.is there any other way than to compile a new ipk? or can you help me compile?

ophub, 2022-07-03:
No need to compile, you change the line in these two files, it says

Roxy09099, 2022-07-03:
> No need to compile, you change the line in these two files, it says
> 
> Download to the local modification, and then overwrite with the same path. Or use vi command to modify online
> 
> ssh: `vi /usr/sbin/openwrt-install-amlogic`

Ok i will try.
let me ask more, here is the code on the wifi chip of the x92 tv box that i have: AP6255 T111102D 1707. can add wifi driver for it?.
[image]

Roxy09099, 2022-07-03:
> vi /usr/sbin/openwrt-install-amlogic

> No need to compile, you change the line in these two files, it says
> 
> Download to the local modification, and then overwrite with the same path. Or use vi command to modify online
> 
> ssh: `vi /usr/sbin/openwrt-install-amlogic`

deleted ?boot but it says I'm in emmc mode

[image]

[image]

ophub, 2022-07-03:
```
df / | tail -n1
lsblk -l
```

Roxy09099, 2022-07-03:
> ```
> # view dtb
> cat /etc/flippy-openwrt-release
> 
> 
> # for wifi deiver:
> dmesg | grep -i sdio
> ls /lib/firmware/brcm/brcmfmac43455-sdio*
> cat /lib/firmware/brcm/brcmfmac43455-sdio.txt
> 
> 
> # foe install to emmc
> df / | tail -n1
> lsblk -l
> ```
> 
> [image]

> ```
> # view dtb
> cat /etc/flippy-openwrt-release
> 
> 
> # for wifi deiver:
> dmesg | grep -i sdio
> ls /lib/firmware/brcm/brcmfmac43455-sdio*
> cat /lib/firmware/brcm/brcmfmac43455-sdio.txt
> 
> 
> # foe install to emmc
> df / | tail -n1
> lsblk -l
> ```
> 
> [image]

thank you i will try. I need to add 2 lines of code:
df / | tail -n1
lsblk -l
enter before exit 1 (marked with a red # by you) or replace the line exit 1?

ophub, 2022-07-03:
The red # is for you to add # in this position. to continue with the installation.

The others are for you to execute it, take screenshots of the execution results one by one, and I will look at your system information

Roxy09099, 2022-07-03:
> ```
> # view dtb
> cat /etc/flippy-openwrt-release
> 
> 
> # for wifi deiver:
> dmesg | grep -i sdio
> ls /lib/firmware/brcm/brcmfmac43455-sdio*
> cat /lib/firmware/brcm/brcmfmac43455-sdio.txt
> 
> 
> # foe install to emmc
> df / | tail -n1
> lsblk -l
> ```
> 
> [image]

* view dtb
[image]

*for wifi driver:
dmesg | grep -i sdio
ls /lib/firmware/brcm/brcmfmac43455-sdio*
[image]

cat /lib/firmware/brcm/brcmfmac43455-sdio.txt
[image]
[image]

ophub, 2022-07-04:
[image]

[image]

What's the difference between you

Roxy09099, 2022-07-04:
> [image]
> 
> [image]
> 
> [#271 (comment)](https://github.com/ophub/amlogic-s9xxx-openwrt/issues/271#issuecomment-1173066971)
> 
> What's the difference between you

before adding the following 2 lines: df / |  tail -n1
 lsblk -l it says I'm in emmc mode, but after adding those 2 lines it shows a Zram line like the picture.  you can review the screenshot
Before:
[image]

After:
[image]

ophub, 2022-07-04:
https://github.com/ophub/amlogic-s9xxx-openwrt/issues/271#issuecomment-1173066971

ophub, 2022-07-04:
The dtb you filled in tx9 is usb tested and can be used? What dtb are you using now?

Roxy09099, 2022-07-04:
> The dtb you filled in tx9 is usb tested and can be used? What dtb are you using now?
> 
> `cat /boot/uEnv.txt`

i am using tx9 pro dtb file because it has wifi driver

ophub, 2022-07-04:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/6aa4a18553dc153ce0ac8516d401e79761298e53/amlogic-s9xxx/common-files/rootfs/etc/model_database.txt#L20

There are options for tx9 in the drop-down menu, you can directly select

Roxy09099, 2022-07-04:
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/6aa4a18553dc153ce0ac8516d401e79761298e53/amlogic-s9xxx/common-files/rootfs/etc/model_database.txt#L20
> 
> There are options for tx9 in the drop-down menu, you can directly select

still not
[image]

ophub, 2022-07-04:
The red # is for you to add # in this position. to continue with the installation.

[image]

Roxy09099, 2022-07-04:
> The red # is for you to add # in this position. to continue with the installation.
> 
> [image]

Roxy09099, 2022-07-04:
> > The red # is for you to add # in this position. to continue with the installation.
> > [image]

I'm sorry I misunderstood you earlier.  i fixed it to # exit 1 and it seems to be working but am stuck at copy boot...

ophub, 2022-07-04:
screenshot

Roxy09099, 2022-07-04:
> screenshot

[image]

ophub, 2022-07-04:
Unplug the power, restart and try again

Roxy09099, 2022-07-04:
> Unplug the power, restart and try again

still fdisk partition fails
[image]
[image]

ophub, 2022-07-04:
The label name of the disk format is incorrect, and it seems that the script needs to be modified in several places.
You can restore the system of the box to Android system.
Do not write emmc, the same is used in USB/TF card.

closed 2022-07-04 by ophub

Roxy09099, 2022-07-04:
> The label name of the disk format is incorrect, and it seems that the script needs to be modified in several places. You can restore the system of the box to Android system. Do not write emmc, the [...]

meson-gxm-q200.dtb and meson-gxm-q200-mod.dtb
What is the difference of these 2 dtb files?  after i changed the dtb file in uEnv.txt to meson-gxm-q200.dtb it was able to write openwrt to emmc.  but after unplugging usb and plugging in power again it doesn't start openwrt but starts android frimware.  how to fix?

ophub, 2022-07-04:
Writing is independent of which dtb is selected.
Your device does not support writing to emmc with the current script. Use it on external devices such as USB/SD/TF.

Roxy09099, 2022-07-04:
> Writing is independent of which dtb is selected. Your device does not support writing to emmc with the current script. Use it on external devices such as USB/SD/TF.
i need your help, I lost my wireless in openwrt. Kernel log:
[   20.179236] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   20.204411] brcmfmac mmc0:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.tanix,tx9-pro.txt failed with error -2
[   20.373717] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   20.436896] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 25 2015 18:58:57 version 7.45.69 (r581703) FWID 01-24037f6e

ophub, 2022-07-04:
Is the firmware you are using now, and the one you used before, a download file?

Roxy09099, 2022-07-04:
> Is the firmware you are using now, and the one you used before, a download file?

Yup

ophub, 2022-07-04:
The last 2 lines of the log you posted above show that you have successfully driven.

Roxy09099, 2022-07-04:
> The last 2 lines of the log you posted above show that you have successfully driven.
> 
> [ 20.373717] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6 [ 20.436896] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 25 2015 18:58:57 version 7.45.69 (r581703) FWID 01-24037f6e

but my wireless card is gone.  wifi is not broken when using android it can still work normally.  I don't understand why it lost wireless so suddenly

ophub, 2022-07-04:
`cat /etc/flippy-openwrt-release`
screenshot

Roxy09099, 2022-07-04:
> `cat /etc/flippy-openwrt-release`
> 
> `cat /lib/firmware/brcm/brcmfmac43455-sdio.txt`
> 
> screenshot
> 
> Screenshot of the welcome interface information of `ssh` (the one with the `banner` logo)

@ophub wireless appeared again in the interface after i flashed fw android 7.0.  there is a problem with my x92 device that makes me want to install openwrt in EMMC, every time I start the device while it is off, I have to press the reset button, otherwise even if the usb is plugged in and i used to start openwrt with  usb before but if the next time I turn it on I don't combine the reset button (in the AV jack) it will boot into android instead of openwrt in usb (tried with fw stock Android 6.0 and 7.0)

ophub, 2022-07-04:
The script does not support your device and writing cannot be used.