ophub/amlogic-s9xxx-openwrt #273: Looks like X96 air P2 is not supported, need help

Roxy09099, 2022-07-08:
i bought a new device, its name is x96 Air P2 (s905x3, ram 4gb, rom 32gb, lan 1000). but when i use kernel version 5.4 & 5.10 and file meson-sm1-x96-air-gbit.dtb. The lan device is recognized as RTL8211F Gigabit Ethernet, and the wifi driver is not installed successfully. when connecting LAN to the computer, the computer cannot recognize the lan ip and display unidentified network.
When I changed to the dtb file of hk1, ugoos x3, h96 max x3 (of course also changed the uboot.ext file corresponding to the dtb file), the wifi driver was installed successfully. lan driver is still RTL8211F Gigabit Ethernet. but when connecting the computer still can't recognize the lan ip and display Unidentified network.
when i use kernel 5.15 and dtb file of HK1 box. everything was working properly. the computer was able to receive the lan ip. can you modify the lan of X96 air P2 to work on kernel 5.10 or 5.4? The fw build I usually use, the author doesn't release with kernel 5.15 .
When I look up information about the x96 air P2 hardware, there are quite a few devices with the same lan and wifi hardware as me being sold in the market. I hope you can support this x96 air P2 version.

pic1:here is the log when i use dtb of Ugoos X3. Driver installed successfully and works fine. however lan have problem as i said above, i tried every dtb file for s905x3 in dtb folder (in kernel 5.4 i don't use uboot.ext , in kernel 5.10 i used uboot.ext file corresponding to file dtb)
[image]

pic2: device's circuit board
[image]

ophub, 2022-07-08:
When you use the dtb file from the HK1 box of 5.15, both wired and wireless work fine?

Roxy09099, 2022-07-08:
> When you use the `meson-sm1-hk1box-vontar-x3.dtb` file from the HK1 box of 5.15, both wired and wireless work fine?

yes, both lan and wireless work

ophub, 2022-07-08:
Several other series of kernels, you use the same configuration, should be able to use

Roxy09099, 2022-07-08:
> Several other series of kernels, you use the same configuration, should be able to use

Hi @ophub  Honestly, I've tried it many times. and i just tried again. same dtb file meson-sm1-hk1box-vontar-x3.dtb but kernel 5.10 lan does not work and 5.15 lan works. Both kernels receive drivers for both lan and wifi in the log. but only 5.15 is active lan. If you don't believe me, I can record a video
[image]

ophub, 2022-07-08:
Different series of kernels have the same driver for hk1, you can compare the source code of the kernel

Roxy09099, 2022-07-08:
> Different series of kernels have the same driver for hk1, you can compare the source code of the kernel
>
> https://github.com/unifreq [...]
Hi @ophub 
in the driver log file still recognizes the lan device as RT8211F (the log file says the driver is still installed successfully) but when connecting to the computer, the computer cannot receive the lan ip and shows the Unidentified network I mentioned The problem is the log file says that the driver has been received but when connecting to the computer via lan. computer does not recognize ip lan with kernel 5.10 and shows Unidentified network. I don't know what's the cause and I'm asking you for a solution, I'm not saying it's completely the driver's fault

ophub, 2022-07-08:
You test it in the usb, update the kernel to 5.4 in the 5.15 kernel: armbian-update 5.4.204

Roxy09099, 2022-07-09:
> You test it in the usb, update the kernel to 5.4 in the 5.15 kernel: `armbian-update 5.4.204`

the problem seems to be with the kernel, after i updated the kernel from 5.15 to 5.4 the lan problem appeared
[image]

ophub, 2022-07-09:
You copy the dtb file that can be used normally in 5.15, put it in 5.4, restart and test it

Roxy09099, 2022-07-09:
> You copy the dtb file that can be used normally in 5.15, put it in 5.4, restart and test it

Lan still doesn't work.
i got a solution.  i updated the kernel from 5.4 to 5.15 and everything works fine.  anyway after installing kernel 5.4 although lan doesn't work but luckily wifi still works, thanks to that i can update to kernel 5.15 successfully

closed 2022-07-09 by Roxy09099